verilog-to-routing/vtr-verilog-to-routing #1849: Multiple Verilog input files

jgoeders, 2021-09-15:
It's my understanding that we don't currently support having multiple Verilog source files.  Is that true?  We talked about having the Edalize runner (https://github.com/olofk/edalize/issues/263) support this.  

I don't believe ODIN II currently supports it.  I wasn't sure if the plan was to:
a) Update ODIN II to support it, or 
b) Update run_vtr_flow.py to accept and concatenate these files before passing to ODIN II>

sdamghan, 2021-09-16:
@jgoeders 
Odin-II currently does support multiple Verilog and Verilog Header files. You would need to pass all files as the same as the following command:
```
./odin_II -V file1.v file2.v file3.vh ...
```
Regarding the `run_vtr_flow` script, new options, named `include_list_add` and `include_dir` are recently added to VTR task config file, so you would be able to pass additional Verilog files to the main design. However, you would need to know that each `include_list_add` file is considered for all `circuit_list_add` files, as the same as the architecture file. For more information I would recommend you to have a look at [hdl_include](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/master/vtr_flow/tasks/regression_tests/vtr_reg_basic/hdl_include) task and [Verilog files](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/master/vtr_flow/benchmarks/hdl_include).

jgoeders, 2021-09-17:
OK, thanks.  Didn't realize this existed.  I'll close this since it sounds like the functionality is there.  Thanks!

closed 2021-09-17 by jgoeders