verilog-to-routing/vtr-verilog-to-routing #1852: Remove redundant vtr_reg_strong run on CI

sfkhalid, 2021-09-16:
Currently the CI runs strong 3 times - twice on github actions (one with the pull request configuration, one with the push configuration, and once on kokoro. The runs on kokoro and github actions push seem to be running strong on the same code (the last commit that was pushed to the branch).

For example, looking at PR #1838, we can see that both the github actions push run and the kokoro run are on commit dff03397 
whereas the github actions pull request run is on commit ca8879.

#### Proposed Behaviour
The proposed behaviour is to stop doing the strong kokoro run, since it is already covered by the github actions strong push run.

#### Possible Solution
Remove the strong.cfg file from the .github/kokoro/presubmit directory

tangxifan, 2021-09-16:
I just double checked. It seems that the VTR strong tests (without sanitizer) are indeed run on both Github Actions and Kokoro.
I do not know the difference between these two runs. If they are the same, we can remove one of them, to save computing power.

QuantamHD, 2021-09-22:
@sfkhalid Just to be clear we want to remove this kokoro configuration. https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/kokoro/presubmit/strong.cfg

Once you confirm I can disable it from the Google side, and then we can remove the file in this repo.

vaughnbetz, 2021-09-23:
You can remove strong.cfg from the continuous runs as well as they are also run by a github runner.  So that is:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/kokoro/continuous/strong.cfg

vaughnbetz, 2021-09-23:
To give a clear spec, there are two configurations to remove:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/kokoro/presubmit/strong.cfg
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/kokoro/continuous/strong.cfg

mithro, 2022-02-24:
I believe this is now done with the removal of Kokoro entirely.

closed 2022-02-24 by mithro