verilog-to-routing/vtr-verilog-to-routing #1909: Using an older version of the 'Catch' unit test framework

Srivat97, 2021-11-10:
Currently, the VTR repo contains version 1 of the 'Catch' unit test framework. Since there is a newer version of 'Catch' (version 2), would it be worthwhile to upgrade?

acomodi, 2021-11-10:
@Srivat97 I see that the current `Catch` framework used in VTR is not part of a subtree as the other `EXTERNAL` libraries (https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/master/libs/EXTERNAL/libcatch).

Following the link in the `README` there is a redirection to the `Catch2` framework that I believe is what you are referring to.

I think it may be worth to rework the `libcatch` library and make it a subtree, and, at this point, upgrade it to the latest version. This might require some adjustments in the current tests.

Srivat97, 2021-11-10:
Since I am updating the documentation for the unit test framework we are using, would it be alright if I went ahead and also upgraded to the latest version as well?

acomodi, 2021-11-10:
Sure, sounds good to me

vaughnbetz, 2021-11-10:
Thanks @acomodi .  Adding @kmurray so we can get his thoughts too, as it looks like he added catch to vtr.  
From a quick look, it looks like catch2 may be  noticably different than catch (multiple .h files vs. one big header file).  It looks like they may have an option to create one big header style still (although it has a different name than before).

So I suggest investigating a bit to see how painful it will be to upgrade before committing to it.

Srivat97, 2021-11-20:
Catch framework has been updated to the latest version in PR #1913.

closed 2021-11-20 by Srivat97