verilog-to-routing/vtr-verilog-to-routing #1921: Switching of RRGraphView

ethanroj23, 2021-11-23:


tangxifan, 2021-11-23:
@ethanroj23 Would you mind providing some code examples? I can better understand the scenario.

ethanroj23, 2021-11-23:
Sure thing. Here are some of the files in question.
[RRGraphViewInterface](https://github.com/ethanroj23/vtr-verilog-to-routing/blob/folded_rr_graph/vpr/src/device/rr_graph_view_interface.h)
[RRGraphView](https://github.com/ethanroj23/vtr-verilog-to-routing/blob/folded_rr_graph/vpr/src/device/rr_graph_view.h)

Specifically, the RRGraphView has a pointer to the interface class that it uses to access functions. 
`RRGraphViewInterface* primary_rr_graph_;`

I think this abstract class pointer may be preventing inlining the function calls that RRGraphView makes.