verilog-to-routing/vtr-verilog-to-routing #1928: Request: Fail if timing is not met

tcal-x, 2021-11-29:
The user should know when their implementation does not meet their stated timing goal.   It is not clear from vpr when this is the case.

#### Proposed Behaviour
Perhaps by default or perhaps only by an option, vpr should have a "fail" exit status when it does not meet the timing constraints.   Also it should not print "VPR succeeded" in that case.  If it is an option, our wrapper can turn it on by default.

#### Current Behaviour
The exit status indicates success, and "VPR succeeded" is printed out, even when 

#### Possible Solution
I think the text above describes things well enough.  Please let me know if I'm mis-characterizing current vpr behavior -- we have wrappers on wrappers so I may be misplacing blame.

Perhaps it makes sense to limit this check to intra-domain paths.

#### Context
We had a user with a design that didn't work on the board.   The root issue was failed timing, but the user was given no indication that this was the problem.

https://github.com/enjoy-digital/litex/issues/1110
https://github.com/SymbiFlow/symbiflow-arch-defs/issues/2331

mithro, 2021-11-29:
I'm sure that we had discussed this previously, but I can't find an existing issue....

tcal-x, 2022-01-12:
@jgoeders , @vaughnbetz , any comment?

The concern is that the user finishes the build with no indication of any problems, and then the design does not run on the board, because it had failed timing badly.

Our workaround would be to have the SymbiFlow wrapper grep the router output for a negative WNS:
```
Final setup Worst Negative Slack (sWNS): -9.92936 ns
```
but this is not a robust solution.   We'd prefer to have the exit status indicated success or failure, where failure includes timing failure.

vaughnbetz, 2022-01-13:
It makes sense to return a different status code or new message to clearly indicate timing failed.  We (and others) have a lot of scripts that might break if we change the status code to be non-zero when execution finished normally, so that solution makes me a bit nervous.  @jgoeders: you know the flow script code better than I; how much would break if we returned a non-zero status for timing failure?  The problem is that many users and tests use: "as fast as possible" as their timing constraint, so while an impossible timing constraint isn't met, the results are meaningful and useful and they want the flow to produce results.

Easier would be to print a clear message saying timing failed, which could be examined by downstream tools, ad wouldn't break anything (for a new message) or much (if we change a message).

tcal-x, 2022-01-13:
How wary are you of adding options?   We could have an option `--fail-status-if-fail-timing` or something like that, that our wrapper would set, but wouldn't change default behavior.

vaughnbetz, 2022-01-13:
I'd be fine with that.

tcal-x, 2022-01-13:
> I'm sure that we had discussed this previously, but I can't find an existing issue....

@mithro there was a similar discussion regarding inconsistent treatment of timing failures in LiteX -- it fails for Lattice flows that don't meet timing (unless you give an argument to ignore timing failure), but with Xilinx flows, timing is not checked.

vaughnbetz, 2022-06-02:
@jmah76 : adding you to this issue as we discussed you taking it as your task list clears.  It would involve:
- add the command line option described above
- add that option to the documentation
- there is a final timing analysis after routing.  If it returned a negative slack anywhere and this option to error out was set, you would call VPR_FATAL_ERROR (stops processing with a message, which should be quite descriptive. I would have to look through the timing analyzer output but I believe we output a worst-negative slack, and also likely give some messages on whether timing requirements were met or not.  You'd look at those values to decide whether or not to throw the error, and if any messages aren't really clear about whether the design met timing or not you'd add in some more explicit ones.

tcal-x, 2022-08-15:
Since #2060 has been merged, I'll close this.   We likely still have some work in F4PGA and CFU Playground to check the vtr result status and handle messaging correctly.

closed 2022-08-15 by tcal-x

vaughnbetz, 2022-08-15:
Great, thanks and let us know if any further changes are needed.