verilog-to-routing/vtr-verilog-to-routing #1935: Warnings in VTR compilation, and No Warnings CI test is not failing

vaughnbetz, 2021-12-10:
There are two issues.

1. We have warnings in the VTR code base when we compile.  Quite a few are in the new catch2 library.  
  - Those should be #pragma'd off (can put the #pragma around the offending header file inclusion) so we don't have spurious warnings (we won't fix these, although we could file an issue on the catch2 github repo for the authors to consider fixing.
  - Other warnings should have issues filed on them, for the appropriate developers to fix if you're not sure how to make them go away.

2. Why didn't the CI test for warnings fail and flag this?  Something seems to be wrong in its settings (we should be compiling with warnings as errors).

@Srivat97 : can you look into this and drive fixes?

Srivat97, 2021-12-11:
1. Will work on this.
2. Looks like we only compile with warnings as errors for libvpr, vpr and ODIN2. I will add the compiler flags to treat warnings as errors for the other projects.

vaughnbetz, 2021-12-16:
@Srivat97 : how is this going?  It sounds like there are multiple moving parts; I think getting the catch2 warnings to go away is the highest priority, as other developers are noticing them in code that used to be warning clean and it could make them miss noticing new warnings introduced by their code.

Srivat97, 2021-12-16:
The warnings introduced by catch2 should now be fixed in PR #1940.

closed 2023-05-12 by vaughnbetz