verilog-to-routing/vtr-verilog-to-routing #1965: Kokoro tests not running reliably

sfkhalid, 2022-02-03:
Some kokoro tests are getting stuck in the pending results stage, or expecting results stage.

#### Expected Behaviour
All of the kokoro tests should run to completion on pull requests.

#### Current Behaviour
On pull requests, most kokoro tests run to completion while one or two get stuck in a pending stage. Different tests get stuck each time.

For example, on PR #1938, all kokoro tests were passing except for the VtR Yosys test. A kokoro force run label was added, which made all runs go to completion except for two different kokoro tests. A kokoro force run label was added again, after which all tests ran to completion except VtR Nightly tests - Part 2 (which had succeeded during the previous two attempts at getting all of the kokoro tests to run).

vaughnbetz, 2022-02-03:
@tangxifan is seeing the same behaviour. He often sees 6 / 9 kokoro tests complete, and has to force a re-run, and then a different subset will team.  It is not consistent in terms of what tests hang.

vaughnbetz, 2022-02-03:
@QuantamHD says kokoro is his domain -- thanks!

vaughnbetz, 2022-02-03:
#1963 may be related.

mithro, 2022-02-03:
Recently we have had some issues with the Google Cloud Platform project which contains the kokoro runners due to some automated log4j checking systems.

@acomodi is looking at trying to move everything to GitHub Action runners hosted on Google Cloud Platform which should be a lot more transparent about when failures.

mithro, 2022-02-24:
With https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1986 - Kokoro has been replaced by GitHub Action Runners and thus this bug is not longer relavent.

closed 2022-02-24 by mithro