verilog-to-routing/vtr-verilog-to-routing #1966: Error: Could not find config/config.txt for task basic_flow

tcal-x, 2022-02-03:
#### Expected Behaviour
Following the build instructions in https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/BUILDING.md and running
```
./vtr_flow/scripts/run_vtr_task.py basic_flow
```
I expect to see
```
k6_N10_memSize16384_memData64_40nm_timing/ch_intrinsics...OK
```

#### Current Behaviour
`Error: Could not find config/config.txt for task basic_flow`

#### Steps to Reproduce
Follow exactly the instructions on https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/BUILDING.md

#### Context
I'm trying to build for the first time in the process of addressing issue #1928.

#### Your Environment

* VTR revision used: current master d15ed6774
* Operating System and version: Ubuntu 20.04
* Compiler version: gcc 9.3.0 is the default `gcc` on my system

ArashAhmadian, 2022-02-07:
@vaughnbetz Sure, I'll make an update to the documentation.

vaughnbetz, 2022-02-07:
Great, thanks Arash.

tcal-x, 2022-02-15:
The updated smoke test works for me now, thanks!

closed 2022-02-15 by tcal-x