verilog-to-routing/vtr-verilog-to-routing #1967: Link to BR96b paper is broken

mithro, 2022-02-06:
The BR96b paper in the https://docs.verilogtorouting.org/en/latest/zreferences/ section is currently broken (it ends in a 404).

```
BR96b
Vaughn Betz and Jonathan Rose. On biased and non-uniform global routing architectures and cad tools for fpgas. CSRI Technical Report 358, University of Toronto, 1996. URL: http://www.eecg.toronto.edu/~vaughn/ papers/techrep.ps.Z.
```

mithro, 2022-02-06:
* http://www.eecg.toronto.edu/~vaughn/%20papers/techrep.ps.Z is a 404 (not found)
* http://www.eecg.toronto.edu/~vaughn/papers/techrep.ps.Z is a 403 (forbidden)

mithro, 2022-02-06:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/81b5aa2a8c54a007046a1d375d934f500bc7471b/doc/src/z_references.bib#L93

michael-etzkorn, 2022-02-16:
> http://www.eecg.toronto.edu/~vaughn/ papers/techrep.ps.Z

Seems like this just needs to remove a space in " papers"
> http://www.eecg.toronto.edu/~vaughn/papers/techrep.ps.Z

umarcor, 2022-02-23:
This can be closed, since #1981 was merged.

closed 2022-02-23 by mithro