verilog-to-routing/vtr-verilog-to-routing #1986: Remove the Google Kokoro runners

mithro, 2022-02-24:
With @acomodi's work in https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1973 we have moved to running the tests under GitHub Action Runners hosted on Google Cloud Platform.

We now need to turn down the Kokoro runners.

mithro, 2022-02-24:
We also need to update the required CI tests in the GitHub repository settings.

mithro, 2022-02-24:
I have removed the Google side configuration files. Kokoro should no longer run on this repository.

mithro, 2022-02-24:
I have removed the Kokoro runs from the required to merge pull request checks.

mithro, 2022-02-24:
@acomodi - Which GitHub Actions runs should replace the previously required Kokoro checks?

acomodi, 2022-02-24:
@mithro `CI Tests` GH action job is replacing the kokoro runs

closed 2025-06-01 by github-actions[bot]