verilog-to-routing/vtr-verilog-to-routing #1994: Running SymbiFlow Benchmarks with 100t and 200t devices

ethanroj23, 2022-03-09:
@acomodi @vaughnbetz @tangxifan I am attempting to run the VTR SymbiFlow (f4pga) benchmarks with the 100t and 200t devices. Currently, I am able to run the benchmarks with the 50t device. I have included the command that I am using to run the 50t device with the linux_arty benchmark.

Does anyone know which files I need to replace in the command? I have downloaded some files for 100t and 200t from SymbiFlow's Google Cloud Platform [here,](https://console.cloud.google.com/storage/browser/symbiflow-arch-defs-gha;tab=objects?pli=1&prefix=&forceOnObjectsSortingFiltering=false) but I haven't been able to successfully run VTR with them yet. Thanks in advance!

`vpr arch.timing.xml 
linux_arty.pre-vpr.eblif 
--read_rr_graph rr_graph_xc7a50t_test.rr_graph.real.bin
--read_router_lookahead rr_graph_xc7a50t_test.lookahead.bin
--read_placement_delay_lookup rr_graph_xc7a50t_test.place_delay.bin 
--max_router_iterations 500 --routing_failure_predictor off --router_high_fanout_threshold 1000 --constant_net_method route --route_chan_width 500 --router_heap bucket --clock_modeling route --place_delta_delay_matrix_calculation_method dijkstra --place_delay_model delta_override --router_lookahead extended_map --check_route quick --strict_checks off --allow_dangling_combinational_nodes on --disable_errors check_unbuffered_edges:check_route --congested_routing_iteration_threshold 0.8 --incremental_reroute_delay_ripup off --base_cost_type delay_normalized_length_bounded --bb_factor 10 --initial_pres_fac 4.0 --check_rr_graph off
--fix_clusters $VTR_ROOT/vtr_flow/benchmarks/symbiflow/place_constr/linux_arty.place
--sdc_file $VTR_ROOT/vtr_flow/benchmarks/symbiflow/sdc/linux_arty.sdc`

acomodi, 2022-03-10:
Hi @ethanroj23, so, to test on either the 100T or 200T you'd need to use the following files from the corresponding tarballs:
- `arch.timing.xml`
- `rr_graph_<part number>.lookahead.bin`
- `rr_graph_<part number>.place_delay.bin`
- `rr_graph_<part number>.rr_graph.real.bin`

There is a catch though using the 200T, and that is that basically the placement constraint file (e.g.  the `--fix_clusters $VTR_ROOT/vtr_flow/benchmarks/symbiflow/place_constr/linux_arty.place` argument) cannot be used, as the pinout for the 200T is different and the constraints specified in the `.place` file are not correct.

I think you can still try using the 100T though, which should have the same pinout (just larger) as the 50T and the constraints should be compatible.

I may create for you some test circuit with other constraints for the 200T, as they are not currently getting produced for the 200T as far as I can see from the tarball generated by F4PGA.

ethanroj23, 2022-03-10:
@acomodi thank you for your response. I will give that a try. If you are able to get a 100T or 200T example running with VTR, please let me know. Thanks!