verilog-to-routing/vtr-verilog-to-routing #1995: Remove some out of date TO-DO and comments in rr_graph.cpp

vaughnbetz, 2022-03-10:
I think there are some out of date comments from the refactoring of the rrgraph apis in rr_graph.cpp (and perhaps related files).

E.g.:            

rr_graph.cpp:
 /* TODO: The casting will be removed when RRGraphBuilder has the following APIs:
             * - set_node_cost_index(RRNodeId, int);
             * - set_node_type(RRNodeId, t_rr_type);
             */

(seems to be done).

rr_graph2.cpp:
 https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/220fa98c059e45549b4ed0227d87f1048b6b0ea4/vpr/src/route/rr_graph2.cpp#L1128
* TODO: these building functions should only talk to a RRGraphBuilder object
 *       The biggest and fatal issue is 
 *       - the rr_graph2.h is included in the rr_graph_storage.h,
 *         which is included in the rr_graph_builder.h
 *         If we include rr_graph_builder.h in rr_graph2.h, this creates a loop
 *         for C++ compiler to identify data structures, which cannot be solved!!!
 *         This will block us when putting the RRGraphBuilder object as an input arguement 
 *         of this function
 */

These should be cleaned up.

tangxifan, 2022-03-10:
@behzadmehmood-rs Can you arrange the team on cleaning up the code comments and create PRs? If you have anything unclear, just let me know. I will review your PR once it is available.

behzadmehmood-rs, 2022-03-11:
> @behzadmehmood-rs Can you arrange the team on cleaning up the code comments and create PRs? If you have anything unclear, just let me know. I will review your PR once it is available.

Sure, we will do that @tangxifan.

vaughnbetz, 2025-05-19:
@amin1377 @soheilshahrouz : these still look like good cleanups to do, although probably best to do after the openFPGA branch merge so we don't create work for ourselves.  Let me know if one of you wants to take a crack at these.