verilog-to-routing/vtr-verilog-to-routing #2017: Download QoR Results Generated During CI Run

amin1377, 2022-04-21:
Is there any documentation on downloading QoR results generated during the CI tests?

Now the only way to access those results is to run the tests again on my local machine.

@tangxifan @vaughnbetz @sdamghan

sdamghan, 2022-04-22:
@amin1377 - once CI tests are completed, you would go to the "checks" tab in a pull request, then click on "Test". Scroll down and you would be able to see artifacts related to all passed tests. By clicking on them you can download each one you like.

I put a sample below which is one of my merged PRs:
1. Click on the "Checks" tab:
[image]

2. Click on "
[image]
Test"

3. Scroll down and click on which artifact you would like to download
[image]

vaughnbetz, 2022-04-23:
Thanks @sdamghan. @amin1377 : can you try this out, and then add this information to the QoR test part of the vtr documentation? We should be quite prescriptive: explain exactly how to do it for the the two standard QoR tests we list (vtr on flagship arch, titan on SIV arch).

amin1377, 2022-04-24:
Sure. I'll add it to the documentaion.

closed 2022-04-29 by amin1377