verilog-to-routing/vtr-verilog-to-routing #2031: Crash when running yosys+odin flow

aman26kbm, 2022-05-10:
I'm running yosys+odin flow on a design. Yosys abruptly ends. And an error is logged in odin.out saying blif file not found.

#### Expected Behaviour

Yosys+odin should finish normally.

#### Current Behaviour

I've attached a zip file that contains the design file. This is an HLS generated design.

I ran two experiments to dig into what's going on.

First, I enabled debug build. With that I see memory leak in the odin log. The odin.out is in the attached zip file.

Second, I tried to see which module in the design is causing the crash.
I ran the flow with all the second level modules under the top module.
All modules passed, except conv_2d_latency_cl_0_0_0_0_0_0_0_0_0_0_0_0_0
Then I ran the flow with all the third level modules under this module.
They all passed as well.
So, the issue is from the logic in this module itself.

I see a giant state machine in this module (1000s of states). May be that's causing the issue.

[bug_may10.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/8664582/bug_may10.zip)

#### Possible Solution

#### Steps to Reproduce

I'm using this architecture: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/COFFE_22nm/k6FracN10LB_mem20K_complexDSP_customSB_22nm.xml 

For debug, I suggest running just this module. You can use a command like this:
../../scripts/run_vtr_flow.py lenet_may10.v ../../arch/COFFE_22nm/k6FracN10LB_mem20K_complexDSP_customSB_22nm.xml --const_gen_inference comb --timing_report_detail aggregated --route_chan_width 300 -top_module conv_2d_latency_cl_0_0_0_0_0_0_0_0_0_0_0_0_0 -temp_dir lenet_may10 -elaborator yosys -fflegalize

#### Context

#### Your Environment

* VTR revision used: master
* Operating System and version:
* Compiler version:

sdamghan, 2022-05-12:
@aman26kbm - it seems the problem is coming from RAM shortage, at least in my system. In the autoname pass, Yosys actually used up more than 34GB of RAM and after that, my system was halted. I suspect the Verilog design results in a giant netlist and Yosys requires a huge amount of RAM to perform the renaming process, specifically for the module mentioned above. The error shown in the Odin-II log file is because the Yosys suddenly ends and the coarse-grained BLIF file will be generated, which results in an abnormal exit for Yosys+Odin-II. Would you be able to try it with more RAM?

aman26kbm, 2022-05-13:
It died at the same point on the larger machine (with 130GB RAM) as well. Looking at the odin log, I see:
Maximum resident set size (kbytes): 121873588
That's about 120 GB. Pretty close to the 130GB RAM size. This was the only thing running on the machine.

On the previous machine (with 58 GB RAM), odin.log contained:
Maximum resident set size (kbytes): 20342420
That's about 20 GB. And there were multiple other runs going on at the same on the machine.

One reason is that this design is so large that it indeed needs a lot of memory. But this design isn't so large compared to a previous design (generated using HLS for the same NN (lenet)) that ran on the smaller machine. I generated this new design for the same NN using HLS with some different pragma settings. In fact, from the point of view of number of lines in the design file, this new design has lesser number of lines than the old one.

But more likely I think there is a memory leak when running yosys. That's causing such a large memory usage. @sdamghan, do you have any ideas/suggestions on debugging this?

sdamghan, 2022-05-13:
I have run only Yosys with the `lenet_may10.v` design file and the same synthesis script of Yosys+Odin-II. The behaviour was the same. The following zip file includes the design file, the synthesis script and required library files.

[lenet_may10_YOSYS_ONLY.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/8688926/lenet_may10_YOSYS_ONLY.zip)

Command to execute Yosys with the synthesis script:
```
> ${VTR_ROOT}/libs/EXTERNAL/libyosys/yosys -c {lenet_may10_YOSYS_ONLY}/synth.tcl
```
---
The memory leak you saw earlier is because the OS abruptly kicked Yosys due to high RAM usage, and then Odin-II exits all processes with failure. There shouldn't be any memory leak in the Yosys+Odin-II and Odin-II codebases since Sanitized and Valgrind CI tests continuously monitor memory leaks in each PR. With that said, I tried to modify the Yosys synthesis script to perform the autoname pass after a few optimization passes. In this way, the renaming process would take considerably lower time and resource consumption since there are fewer things to rename. Yosys executed the autoname pass successfully with this change and there was no RAM shortage anymore. I will make a new PR to move the autoname pass permanently after the optimization passes, however, I need to make sure it does not affect the coarse-grained netlists generated by Yosys because it might cause further failures in the Yosys+Odin-II flow. 

As a temporary solution to this issue, I would suggest applying the following change to the `YYosys.cpp` file:

```
262| // Use a readable name convention
263| run_pass(std::string("autoname"));
264| // Translate processes to netlist components such as MUXs, FFs and latches
265| run_pass(std::string("proc; opt;"));
266| // Extraction and optimization of finite state machines
267| run_pass(std::string("fsm; opt;"));
268| // Collects memories, their port and create multiport memory cells
269| run_pass(std::string("memory_collect; memory_dff; opt;"));
```
to
```
262| // Translate processes to netlist components such as MUXs, FFs and latches
263| run_pass(std::string("proc; opt;"));
264| // Extraction and optimization of finite state machines
265| run_pass(std::string("fsm; opt;"));
266| // Collects memories, their port and create multiport memory cells
267| run_pass(std::string("memory_collect; memory_dff; opt;"));
268| // Use a readable name convention
269| run_pass(std::string("autoname"));
```
Hope this helps to remove the current barrier :), and please let me know if you need further help

aman26kbm, 2022-05-13:
Gotcha. I'll update the YYosys.cpp file and go ahead with that for now.

sdamghan, 2022-05-20:
resolved in #2035

closed 2022-05-20 by sdamghan