verilog-to-routing/vtr-verilog-to-routing #2041: Github CI tests are doubled

MohamedElgammal, 2022-05-26:
Every CI test on GitHub actions is running twice:

- One triggering push
- One triggering Pull request

Actually, both runs every time anyone pushes to a branch on a PR. 

#### Current Behaviour

Currently, some tests are failing on the pull request trigger while the exact same test is passing for the push trigger.

The error is:
`
Could not find rsyslog for current run!

ERROR: (gcloud.beta.compute.instances.create) Could not fetch resource:
- The zone 'projects/***/zones/us-west1-a' does not have enough resources available to fulfill the request.  '(resource type:compute)'.
`
#### Possible Solution

Remove actions on pull_request trigger action

MohamedElgammal, 2022-05-26:
@kgugala one example that is currently happening is the failing tests on PR #2034

mithro, 2022-05-26:
@MohamedElgammal - The general idea is that people shouldn't be pushing directly to the `vtr-verilog-to-routing` but to their own fork of the repository and then sending a pull request from that....

https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/getting-started/about-collaborative-development-models#fork-and-pull-model
https://gist.github.com/Chaser324/ce0505fbed06b947d962
https://reflectoring.io/github-fork-and-pull/

MohamedElgammal, 2022-05-26:
@mithro Thanks for sharing this with me . I will make sure to follow this model in my upcoming pull requests.

MohamedElgammal, 2022-05-26:
However, I think we still have some issues regarding the available resources on the cloud as we are having too many failing tests.

mithro, 2022-05-26:
That should be a seperate GitHub issue.

MohamedElgammal, 2022-05-26:
Thanks. Closing this issue and opening #2042  to discuss the failing tests due to cloud resources.