verilog-to-routing/vtr-verilog-to-routing #2059: VTR flow library documentation is just a skeleton -- should fill in

vaughnbetz, 2022-06-09:
https://docs.verilogtorouting.org/en/latest/vtr/python_libs/vtr/ has the skeleton of documentation for this library, but only the headings are there.

#### Proposed Behaviour
Complete the documentation

#### Context
Our flow scripts are pretty complex (and useful) so documenting them is helpful. We do have pretty reasonable documentation on the overall scripts, so if you conclude no documentation is needed on the library you could instead delete this documentation tab.  But I suspect documentation here will be useful.

vaughnbetz, 2025-05-19:
@jgoeders : we should close this one out. We still have just a skeleton of documentation for the VTR flow library. If you or a delegate can fill it out some that would be best.

jgoeders, 2025-05-19:
@WhiteNinjaZ Could you work on documenting this?

WhiteNinjaZ, 2025-05-20:
Yea I can fill this in

vaughnbetz, 2025-05-20:
The CI automatically updates the docs.  Not sure why they don't show up then ... I'm guessing there is something wrong with the doxyfile that says which ones to show on the web.  See https://docs.verilogtorouting.org/en/latest/dev/code_documentation/
Possibly it is different for python ... our documentation on doxyfiles seems to be C++ focused.  If you determine we should update the developer guide doxyfile section go ahead and do that too.

vaughnbetz, 2025-05-20:
In case what I wrote is confusing: to make the web docs we need to do two things:
- Actually write the comments in Doxygen format in the source files
- Modify the doxyfiles (.rst files) as specified in https://docs.verilogtorouting.org/en/latest/dev/code_documentation/#validation-of-doxygen-comments-updating-api-reference
- CI will build the docs, and clicking on the documentation link in the CI tests is simpler than building locally, so that's what I usually do.

WhiteNinjaZ, 2025-05-21:
@jgoeders

PR #3080 (2025-05-23): Fixing documentation build

---

#### Description
Readthedocs was not correctly running sphinx autodoc on our python modules because of broken imports.  We were instructing readthedocs to install the `docs/requirements.txt`, but not the base `requirements.txt`.  The latter includes packages that are used by our Python scripts.  autodoc cannot document the Python files without having the necessary libraries available.

#### Related Issue

Fixes #2059 

#### Motivation and Context

#### How Has This Been Tested?

#### Types of changes

- [x] Bug fix (change which fixes an issue)
- [ ] Breaking c [... cut]
closed 2025-05-23 by jgoeders