verilog-to-routing/vtr-verilog-to-routing #2088: VPR Pack: Replace hand-implemented linked lists to C++'s std::lists

jmah76, 2022-07-07:
Related to PR #2084 

Lists such as the ones vpr/src/base/vtr_types.cpp (line 236) might be better implemented as std::lists. Refer to https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2084#discussion_r915061111

vaughnbetz, 2022-09-06:
@jmah76 : were these mostly / all fixed (in which case we should close this issue), or is keeping it around worthwhile?

vaughnbetz, 2025-05-19:
Sending this to @AmirhosseinPoolad .... updae some of the clustering data structs to use std::vector and std::list etc. Some have already been done, but more remain.

AmirhosseinPoolad, 2025-05-26:
I ran clang-tidy to give me a list of all C-style arrays used in VPR. It's missing the heap allocated pointer arrays, still trying to find out if there's an easy way to list those. You can find results below. 

https://gist.github.com/AmirhosseinPoolad/bee1015b11f165310e0861f5d81d54d7