verilog-to-routing/vtr-verilog-to-routing #2094: CI failures

sdamghan, 2022-07-14:
It seems the CI failure is coming from my merged PR (#2081). Previously, when we modify GitHub action configuration files, the changes should be merged into the master branch to become effective. However, this time seems like it resulted in failures.

@vaughnbetz - I will have a look at that to find a solution.

#### Expected Behaviour

#### Current Behaviour

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

vaughnbetz, 2022-07-14:
Thanks.  If the CI failures can't be resolved quickly we should revert the changes in #2081 while we figure them out.

sdamghan, 2022-07-14:
Sure

sdamghan, 2022-07-15:
The compile flag was missed in other CI tests of vtr_reg_basic. I close this issue as it's fixed by #2095

closed 2022-07-15 by sdamghan