verilog-to-routing/vtr-verilog-to-routing #2103: Yosys+ODIN flow crashing when ODIN only flow passes in some benchmarks

aman26kbm, 2022-07-23:
#### Expected Behaviour
The Yosys+Odin flow should pass as well.

#### Current Behaviour
I'm seeing a few benchmarks run successfully with ODIN only, but they crash with ODIN+Yosys.

The benchmarks are:
1. [vtr-verilog-to-routing/eltwise_layer.v at master · aman26kbm/vtr-verilog-to-routing (github.com)](https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/benchmarks/verilog/koios/eltwise_layer.v)
2. [vtr-verilog-to-routing/softmax.v at master · aman26kbm/vtr-verilog-to-routing (github.com)](https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/benchmarks/verilog/koios/softmax.v)
3. [vtr-verilog-to-routing/attention_layer.v at master · aman26kbm/vtr-verilog-to-routing (github.com)](https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/benchmarks/verilog/koios/attention_layer.v)

The arch file is:
[vtr-verilog-to-routing/k6FracN10LB_mem20K_complexDSP_customSB_22nm.xml at master · aman26kbm/vtr-verilog-to-routing (github.com)](https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/arch/COFFE_22nm/k6FracN10LB_mem20K_complexDSP_customSB_22nm.xml)

The config file (for one of the designs) is:
[vtr-verilog-to-routing/config.txt at master · aman26kbm/vtr-verilog-to-routing (github.com)](https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/tasks/koios/exp1a_yosodin_hard/agilex.attention_layer/config/config.txt)
For other designs, a similar config file is used.

Note that I'm including the hard block include file ([vtr-verilog-to-routing/hard_block_include.v at master · aman26kbm/vtr-verilog-to-routing (github.com)](https://github.com/aman26kbm/vtr-verilog-to-routing/blob/master/vtr_flow/benchmarks/verilog/koios/hard_block_include.v)) to enable hard blocks.

On running with debug, I didn't get much. Here's the stack trace:

 ```
76  96%|================================================>--| Remaining: 987.7ms
 77  98%|=================================================>-| Remaining: 487.0ms
 78 100%|==================================================>| Total time: 24022.9ms
 79 -------------------------------------
 80 /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/adders.cpp:933:38: runtime error: member access within null pointer of type 'struct nnode_t    '
 81 AddressSanitizer:DEADLYSIGNAL
 82 =================================================================
 83 ==23287==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000020 (pc 0x5605db211dd2 bp 0x7ffeb684a100 sp 0x7ffeb684a0b0 T0)
 84 ==23287==The signal is caused by a READ memory access.
 85 ==23287==Hint: address points to the zero page.
 86     #0 0x5605db211dd1 in match_node(vtr::t_linked_vptr*, operation_list) /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/adders.cpp:933
 87     #1 0x5605db211aaa in traverse_list(operation_list, vtr::t_linked_vptr*) /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/adders.cpp:911
 88     #2 0x5605db211a00 in reduce_operations(netlist_t*, operation_list) /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/adders.cpp:901
 89     #3 0x5605dafcbe8f in optimization /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/odin_ii.cpp:163
 90     #4 0x5605dafcc61f in synthesize /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/odin_ii.cpp:250
 91     #5 0x5605dafcdb15 in start_odin_ii(int, char**) /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/odin_ii.cpp:354
 92     #6 0x5605dafc910b in main /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/main.cpp:30
 93     #7 0x7f4f37fcdbf6 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21bf6)
 94     #8 0x5605daebc819 in _start (/export/aman/vtr_aman/vtr-verilog-to-routing/build/ODIN_II/odin_II+0xd47819)
 95
 96 AddressSanitizer can not provide additional info.
 97 SUMMARY: AddressSanitizer: SEGV /export/aman/vtr_aman/vtr-verilog-to-routing/ODIN_II/SRC/adders.cpp:933 in match_node(vtr::t_linked_vptr*, operation    _list)
 98 ==23287==ABORTING
 99 Command exited with non-zero status 1
```

#### Possible Solution

#### Steps to Reproduce

Please use the design, arch and config files linked above

#### Context

This is an issue encountered while working on Koios 2.0 benchmark suite.

#### Your Environment

* VTR revision used:
* Operating System and version:
* Compiler version:

aman26kbm, 2022-07-24:
Hey Seyed,

We've been able to identify the root cause of this issue. It happened when we got the same error in another benchmark (brainwave).

This crash is happening because there are multiple pb_types for the same model (subckt) in the arch file.
For example, see 'multiply_fp_clk' in the arch file linked above. This is the name of the model (subckt).

```
 242     <model name="multiply_fp_clk">
 243       <input_ports>
 244         <port name="clk" is_clock="1"/>
 245         <port name="a" clock="clk" combinational_sink_ports="out"/>
 246         <port name="b" clock="clk" combinational_sink_ports="out"/>
 247       </input_ports>
 248       <output_ports>
 249         <port name="out" clock="clk"/>
 250       </output_ports>
 251     </model>
```

But there are two pb_types that use this subckt with different port widths.

First one (with 32-bit ports):
```
1867           <pb_type name="mult_fp32" blif_model=".subckt multiply_fp_clk" num_pb="1">
1868             <input name="a" num_pins="32"/>
1869             <input name="b" num_pins="32"/>
1870             <output name="out" num_pins="32"/>
1871             <clock name="clk" num_pins="1"/>
```

Second one (with 16-bit ports):
```
1897           <pb_type name="mult_fp16" blif_model=".subckt multiply_fp_clk" num_pb="2">
1898             <input name="a" num_pins="16"/>
1899             <input name="b" num_pins="16"/>
1900             <output name="out" num_pins="16"/>
1901             <clock name="clk" num_pins="1"/>
```

In the ODIN only flow, this was okay. ODIN would figure out which pb_type to use based on port widths. 
But in the Yosys+ODIN flow, there is a problem. The flow generates black box definitions of hard blocks in arch_dsp.v.
In this file, these models/pb_types show up as verilog modules. Now, I see that there is only one verilog module for multiply_fp_clk. See below:

226 (* blackbox *)
227 module  multiply_fp_clk(
228   input [32:0]  b,
229   input [32:0]  a,
230   input [1:0] clk,
231   output  [32:0]  out
232 );
233 /* the body of the hardblock model is empty since it should be seen as a blackbox */
234 endmodule

There is no matching module for the multiply_fp_clk pb_type with 16-bit ports. This causes the failure.

A workaround here is to have a one-to-one correspondence between pb_type and model/subckt, for hard blocks. 
This needs to change the arch file. I can do that, but please let me know what you think. If the issue itself is easy to fix, then we don't have to use the workaround. Otherwise, we can go ahead with the workaround for now.

Tagging @saitama0300, FYI.

sdamghan, 2022-07-24:
Hi Aman,

That makes sense. In Odin-II, we only extract the name of hardbacks and the port orders from the architecture file. In this way, Odin-II handles the port sizes, considering the upper limit value of hard block ports, defined in the architecture model. However, the scenario for Yosys+Odin-II is a bit different, as we require to provide Yosys with the declaration of hard blocks to let Yosys know how to behave them.

As you mentioned, one approach is to define a new HB in the architecture file with matching port sizes. Another approach would be to define a width parameter for hard blocks and ask users to define the width of the ports with the instantiation (similar to memory blocks). An important note that is worth mentioning is that, we need to determined parameters from the architecture file. The final hard blocks declaration, which will be provided to Yosys would be like the below:
```
226 (* keep_hierarchy *)
227 module  multiply_fp_clk
228 # (
229   parameter WIDTH = 1
230 )(
231   input [`WIDTH-1:0]  b,
232   input [`WIDTH-1:0]  a,
233   input [1:0] clk,
234   output  [`WIDTH-1:0]  out
235 );
236 /* the body of the hardblock model is empty since it should be seen as a blackbox */
237 endmodule
```
Then, the instantiation also would be something like the below:
```
...
defparam mult_inst.WIDTH = 16;
multiply_fp_clk mult_inst (.a(a), .b(b), .clk(clk), .out(out));
...
```

In the second approach, Yosys does not receive the hard blocks as blackboxes. On the other hand, and like memory blocks in the VTR flow, Yosys generates modules of _multiply_fp_clk_ with different port sizes and consider them as separate modules using its hash naming. Odin-II also knows how to properly interpret these hashed _.models_ in the coarse-grained BLIF file.

I will consider this as an enhancement to the Yosys+Odin-II synthesis flow to cover the previous behaviour of Odin-II. Meanwhile, I would recommend you and @saitama0300 move forward and provide a new definition for the above-mentioned hard block with the matching port size with your instantiation for now, as the development process might take time.

Adding @poname to this thread, as this corner case should be properly handled in the Odintechmap as well as the current version of Yosys+Odin-II.

aman26kbm, 2022-07-24:
Thanks, Seyed. 

So, you mean we need to go ahead with the workaround for now, right? The parameter based solution you mention (similar to block rams) is the correct solution that will take time to implement, right? I ask to clarify because for this solution, the fix in the design files is simple. 

Also, I tried the workaround, but I still see the crash. That's surprising. Maybe I'm missing something. I am attaching a tar ball, with arch file, design file and command. Note that the command uses the `-top_module` switch. This reduces the run time to a few seconds.

Can you take a look at it and see what's going on?

[for_seyed_issue2103.tar.gz](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/9176249/for_seyed_issue2103.tar.gz)

aman26kbm, 2022-07-24:
Oh adding to the previous comment.. 
The design I attached uses multiply_fp_clk and adder_fp_clk.
I separated the models in the arch file - multiply_fp_clk and multiply_fp_clk_32, adder_fp_clk and adder_fp_clk_32.

sdamghan, 2022-07-24:
@aman26kbm - yes, please move forward with the workaround. 

Regarding the aforementioned approaches, we cannot definitely determine which approach is correct and which one is wrong since if we are supposed to consider hard blocks as black boxes in the synthesis process, the synthesizer should not perform any modification on the HB port sizes, as Yosys does in this way. Once a user defines a module as a black box, Yosys only considers the module definition and even by explicitly specifying the parameters, their values do not change. 

In Odin-II, this feature was embedded from the beginning as the tool was specifically designed for the VTR flow to read architecture files and parses HDL files accordingly. 

P.S: I do not have access to my system from home; I will take a look at the designs tomorrow.

sdamghan, 2022-07-25:
Hey Aman,

Long story short, the issue comes from the DSP block's name. Using Yosys as the elaborator, it generates names for subcircuits that have the name of the subcircuit and some information about its port sizes in very specific cases, like `dual_port_ram\\ADDR_WIDTH?4'0100\\DATA_WIDTH?4'0101`.

In the Odin-II BLIFReading phase, we require to extract the subcircuit types from their name. Therefore, custom hard blocks should not **INCLUDE** any of Yosys subcircuit names in addition to the VTR primitive blocks. Simply, whatever is defined below as the `yosys_subckt_strmap` is prohibited to be included in the custom DSP block naming. https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/0c715b66c62227f66399bd74d76353a39cefeba6/ODIN_II/SRC/enum_str.cpp#L257-L488

Discussing the issue, the DSP block included the word `multiply` in its name, which is the name of VTR multiply primitive. Therefore, in the coarse-grained BLIF elaboration phase, the subcircuit was considered a MULTIPLY hard block instead of a HARD IP. Because the multiply defined in the architecture has the upper limit size of 27 for its input ports, a few input pins of the `multiply_fp_clk` instance got freed before when they were supposed to, and we saw segfault.

***Simple solution***: changing `multiply_fp_clk` to `mult_fp_clk`

aman26kbm, 2022-07-25:
:) Gotcha. I'll update the arch file. Thanks, @sdamghan

aman26kbm, 2022-07-25:
Hey Seyed,
In the arch file we have, I see that we have many models which are specified in the blif_model of various pb_types. For example, mac_int is a model that is used in the blif_model of 3 pb_types. So, for this workaround, I'll have to add separate models for each. 

That's not hard, but I just wanted to something about the non-workaround method (just to understand this correctly).
In the non-workaround method, we will instantiate a hard block and specify it's parameter like you mentioned earlier (pasted below):
 
```
defparam mult_inst.WIDTH = 16;
multiply_fp_clk mult_inst (.a(a), .b(b), .clk(clk), .out(out));
```
 
Where do we define the "WIDTH" parameter? Is that somewhere in the ODIN code? Or is it defined in the arch file? 
BRAMs have ADDR_WIDTH and DATA_WIDTH. I don't see them in the arch file or in the ODIN codecase.

sdamghan, 2022-07-26:
@aman26kbm - correct. That's why I mentioned that please move forward with the workaround. We need to come up with an appropriate design for this case.
As mentioned earlier, Odin-II was designed to handle these cases for the VTR flow. So we cannot accommodate this behaviour in the Arch file since other tools are also reading arch files, and we only need this case to be handled for Yosys+Odin-II. Second, there is no HDL declaration of HBs in the arch file. 
We could handle such cases for the memory hard blocks since we have a unique and specified definition for them in `vtr_flow/primitives.v`. However, the scenario is not the same for user-defined custom hard blocks.
To overcome this issue, we can take two approaches. The first is to ask users to provide the declaration of hard blocks in addition to the arch file. The second would be handling this when we extract the declaration of hard blocks from the arch file (i.e. in `arch_dsp.v`).

sdamghan, 2022-07-26:
Please do not close this issue as its a reminder to propose the implementation

aman26kbm, 2022-07-26:
Thanks for the explanation. Sure, will not close it. For now, we have a workaround. We can use this issue to track the work for the correct fix.

sdamghan, 2022-07-26:
The libarchfpga provides a list of architecture models in the `t_arch/models` data structure. Regardless of complex block modes and the number of physical blocks they include, the input size of a complex hard block (blif_model) is considered the largest among all modes. In the synthesis process, whether in Odin-II or Yosys+Odin-II, the model considered for a hard block is what libarchfpga returns, which is the largest one as mentioned above.

For instance, in the `mult_fp_clk` case, as the 32 bits mode has the largest width, the model returned by libarchfpga has 32 bits width. Therefore, the final netlist has hard block models with 32 bits signal width, regardless of the width of the connected signal to a complex block instance. If connected signals have a width less than 32 bits, the synthesizer will extend them using pad pins. Other than that, if larger than 32 bits, the additional pins will not be connected to the hard block.

Closing this issue as there is no problem with the codebase.

closed 2022-07-26 by sdamghan