verilog-to-routing/vtr-verilog-to-routing #2123: Support for simple dual port RAMs in ODIN/ODIN+Yosys

aman26kbm, 2022-08-11:
Moving from email to GitHub feature request.

#### Proposed Behaviour
Provide support for simple dual port (SDP) RAMs in ODIN/ODIN+Yosys
[image]

#### Current Behaviour
Currently, only single port (SP) and true dual port (TDP) RAMs are supported:

```
 
 *                                              en2                         *                                                             
 *                                    en1      |                            *                                           en                                
 *                       CLK       |          |                            *                     CLK                 |                                   
 *                      __|_____|_____|___                      *                      __|_________|_____                          
 *                     |                  |                      *                     |                                 |                         
 *                     |                  |                      *                     |                                 |                          
 *     data1 --->|                  | <--- addr1     *                     |                                 |                         
 *                     |          DPRAM     |                      *        data --->|          SPRAM          |<--- addr              
 *     data2 --->|                                |<--- addr2      *                      |                                 |                         
 *                     |                                |                      *                     |                                  |                         
 *                     |________________|                      *                     |_________________|                         
 *                              |            |                                 *                                        |                                   
 *                              |            |                                 *                                        |                                   
 *                              v           v                                *                                        v                                    
 *                           out1        out2                           *                                       out                                  
        
```

#### Possible Solution

#### Context

This is not needed right now. This is for the future. Adding Dr. Kenneth Kent to assign it to a future student.

mithro, 2022-08-21:
FYI - Yosys recently went through a pretty big refactor to support more types of memory styles. See the work from 
@mwkmwkmwk - https://github.com/YosysHQ/yosys/pulls?q=author%3Amwkmwkmwk+

sdamghan, 2022-08-26:
@poname - considering the Yosys recent changes on memory blocks, you would be required to consider upstreaming the VTR-Yosys for Odintechmap and provide the support for newly added components, as well. This could be postponed to the second phase of your development as we expect Odintechmap initially work fine with the old version of Yosys added to the VTR repository.
Adding @vaughnbetz on this thread as he already requested for upstreaming VTR-Yosys.

vaughnbetz, 2025-05-19:
Not sure if the move to yosys + parmys has resolved this, but sending to Logan to check once he has time. @loglav03 : let me know if you need more info on this. I suggest trying a test case once you have the latest yosys going, to see if we use simple dual port RAMs (one read, one write port) when we can, instead of using true dual port (two R/W ports).

loglav03, 2025-06-16:
@vaughnbetz - Now that the latest Yosys is working with vtr, I'm ready to start testing this out. I'm not to sure what commands I should be using for this (circuit and architecture) and I'm assuming I should be looking at some pre-vpr output file

aman26kbm, 2025-06-17:
Yeah I don't think we have an arch file with simple dual port RAMs. Vaughn's model looks good. 

In the pb_type, we have something called port_class. For single port, we have:

          <input name="addr" num_pins="14" port_class="address"/>
          <input name="data" num_pins="2" port_class="data_in"/>
          <input name="we" num_pins="1" port_class="write_en"/>
          <output name="out" num_pins="2" port_class="data_out"/>
          <clock name="clk" num_pins="1" port_class="clock"/>

And for true dual port, we say:

          <input name="addr1" num_pins="10" port_class="address1"/>
          <input name="addr2" num_pins="10" port_class="address2"/>
          <input name="data1" num_pins="32" port_class="data_in1"/>
          <input name="data2" num_pins="32" port_class="data_in2"/>
          <input name="we1" num_pins="1" port_class="write_en1"/>
          <input name="we2" num_pins="1" port_class="write_en2"/>
          <output name="out1" num_pins="32" port_class="data_out1"/>
          <output name="out2" num_pins="32" port_class="data_out2"/>
          <clock name="clk" num_pins="1" port_class="clock"/> 

I don't know where the port class is used, but we will need to have address1, address2 and then data_in and data_out for simple dual port RAM.