verilog-to-routing/vtr-verilog-to-routing #2152: Issue with running Yosys flow

ept221, 2022-09-05:
There seems to be an issue with running the basic usage example for `run_vtr_flow` with `-start yosys`.

#### Expected Behaviour
In the [docs](https://docs.verilogtorouting.org/en/latest/yosys/quickstart/), the example invocation for running VTR with Yosys as the front-end is: `./run_vtr_flow PATH_TO_VERILOG_FILE.v PATH_TO_ARCH_FILE.xml -start yosys`. This should run without error.

#### Current Behaviour
Instead, when that command is invoked, Python produces a traceback to an error: 
```
Traceback (most recent call last):
  File "../vtr_flow/scripts/run_vtr_flow.py", line 807, in <module>
    sys.exit(vtr_command_main(sys.argv[1:], prog=sys.argv[0]))
  File "../vtr_flow/scripts/run_vtr_flow.py", line 546, in vtr_command_main
    vtr.run(
  File "/home/ept/tools/vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/flow.py", line 213, in run
    vtr.yosys.run(
  File "/home/ept/tools/vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/yosys/yosys.py", line 191, in run
    shutil.copyfile(yosys_base_models, yosys_models_full_path)
  File "/home/ept/miniconda3/envs/vtr/lib/python3.8/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/home/ept/tools/vtr-verilog-to-routing/vtr_flow/scripts/temp/yosys_models.v'
```
The issue appears to be in [vtr_flow/scripts/python_libs/vtr/yosys/yosys.py](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/scripts/python_libs/vtr/yosys/yosys.py). Beginning at line 190, there appear to be various paths with a name: *_full_path. I believe these are paths are for copying files into the `temp_dir` that gets created during this flow. However, appended to many of these paths is: `vtr.paths.scripts_path` which I *believe* is a mistake. In my case, printing `vtr.paths.scripts_path` out yields: `/home/ept/tools/vtr-verilog-to-routing/vtr_flow/scripts`. So it looks like the script is trying to copy files into `/home/ept/tools/vtr-verilog-to-routing/vtr_flow/scripts/temp` instead of `my_invocation_dir/temp/` (where `my_invocation_dir` is the directory where I invoked vtr_flow).

#### Possible Solution
I think this should be as easy as removing `vtr.paths.scripts_path` from the beginning of the target paths. I would have created a PR, but wasn't sure I had this right.

#### Steps to Reproduce

1. `cd $VTR_ROOT`
2. `mkdir stuff`
3. `cd stuff`
4. `../vtr_flow/scripts/run_vtr_flow.py $VTR_ROOT/doc/src/quickstart/blink.v ../vtr_flow/arch/timing/EArch.xml -start yosys`

#### Context

#### Your Environment

* VTR revision used: `2f34458`
* Operating System and version: Ubuntu 20.04.4 LTS 
* Compiler version: gcc version 9.4.0

ept221, 2023-01-22:
One workaround to this is to specify a full path to the temp dir with  `-temp_dir`. For example: `./run_vtr_flow PATH_TO_VERILOG_FILE.v PATH_TO_ARCH_FILE.xml -start yosys -temp_dir /full/path/to/temp_dir`

farukyld, 2024-03-26:
I am also able to resolve this by replacing this line
```python
    architecture_file_path = str(vtr.paths.scripts_path / architecture_file)
```
with this:
```python
architecture_file_path = str(architecture_file)
```
in this file: `vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/parmys/parmys.py` line `212` at this commit:

#23535b02bb3afde26daf9d3887606c9e21192311