verilog-to-routing/vtr-verilog-to-routing #2154: Splitting pin locations for a tile with capacity >1

ganeshgore, 2022-09-07:
In the architecture I am working with, the grid_io is modeled as follows.
In this particular layout, the IOs are placed in the grid instead of periphery.
How can I split the `a2f_o` output pin locations to the left, right, and bottom?
a2f_o signal is single bit, so something like io_top.a2f_o[0:2] does not work

**IO_TILE**
```
<tile name="io_top" capacity="22" area="0">
    <equivalent_sites>
    <site pb_type="io"/>
    </equivalent_sites>
    <clock name="clk" num_pins="8"/>
    <input name="f2a_i" num_pins="1"/>
    <output name="a2f_o" num_pins="1"/>
    <input name="sc_in" num_pins="1"/>
    <output name="sc_out" num_pins="1"/>
    <input name="reset" num_pins="1" is_non_clock_global="true"/>
    <fc in_type="frac" in_val="0.25" out_type="frac" out_val="0.5">
    <fc_override port_name="clk" fc_type="frac" fc_val="0"/>
    <fc_override port_name="sc_in" fc_type="frac" fc_val="0"/>
    <fc_override port_name="sc_out" fc_type="frac" fc_val="0"/>
    <fc_override port_name="reset" fc_type="frac" fc_val="0"/>
    </fc>
    <pinlocations pattern="custom">
    <loc side="top">io_top.a2f_o io_top.f2a_i io_top.clk io_top.sc_in io_top.sc_out io_top.reset</loc>
    </pinlocations>
</tile>

```
#### Expected Behaviour

Options to declare pin locations when 

#### Current Behaviour

I tried io_top.a2f_o[0:2] which give me this error
```
LoadPinLoc: Assertion 'pin_num < type->num_pins / type->capacity' failed.
```

and for  io_top[0:2].a2f_o i get following error
```
Pin location specification 'io_top[12:0].a2f_o' should not contain an instance range (should only be the block name)
Error 2: Command 'vpr' execution has fatal errors
```

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used: VTR 8
* Operating System and version:
* Compiler version:

vaughnbetz, 2022-09-08:
@kmurray @tangxifan : I think we can't specify this syntax right now. @ganeshgore wants to connect different IO pins to different routing channels within this group of 22 in a tile. 
This seems like a reasonable extension for the rr-graph builder; allow the different sub-items within this tile to specify different connectivity to the channels.
@ganeshgore also points out that he can't refer to the sub-items within this tile for direct connections either.

tangxifan, 2022-09-08:
@vaughnbetz I think 

- First, this is a missing syntax supported by the architecture XML parser. 
- Second, for the direct connections, I remember the detailed pin assignments are done inside function ``build_rr_graph()``. Perhaps we should extend the codes there, in addition to ``libarchfpga``

Since @ganeshgore needs the feature currently when using OpenFPGA, my plan is to

- Implement the syntax support in the current VPR of OpenFPGA. Test and ensure it is working fine using regression tests.
- Migrate the enhancements from OpenFPGA to the feature branch ``openfpga`` of VTR. Then, we can start creating PRs based on your preference (since there are a lot of changes in the feature branch already, it is better to split into a few small PRs).

Let me know what you think.

vaughnbetz, 2022-09-08:
Thanks @tangxifan .  That makes sense and I think this is a good feature.

tangxifan, 2022-09-09:
The issues is addressed in OpenFPGA's VPR: https://github.com/lnis-uofu/OpenFPGA/pull/775

I have added the codes to the feature branch ``openfpga``. See details in #2135 
Will discuss about how to merge this specific feature to master branch soon.

closed 2022-09-10 by ganeshgore