verilog-to-routing/vtr-verilog-to-routing #2169: Long packing time of CI tests

saaramahmoudi, 2022-10-13:
vtr_reg_nightly_test1 is cancelled due to a long file packing time. Tried to run test multiple times and It got cancelled every time due to the timeout. This issue is made upon @vaughnbetz request.

#### Expected Behaviour
Test should be passed as it has no qor test failures and no run failures.

#### Current Behaviour
It printed "All tests passed" and went to "Packing benchmarks files", then cancelled due to the timeout.

#### Link to a PR that produces the error
[Wirelength Estimation PR](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2168)

vaughnbetz, 2022-10-13:
@kgugala @mithro

vaughnbetz, 2022-10-13:
@zhaisitong : has a workaround of moving some tests from nightly1 to nightly5.
He's going merge that sooner rather than later (it's in a WIP PR right now).
But we should still sort out why the pack time is so high.

kgugala, 2022-10-17:
@saaramahmoudi can you link to a failed CI run? The PR you linked is green

saaramahmoudi, 2022-10-17:
Sure, attached. @kgugala 
[First Try](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/3227964548/attempts/1)
[Second Try](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/3227964548/attempts/2)
[ThirdTry](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/3227964548/attempts/3)

kgugala, 2022-10-17:
hmm, this seems to be VPR issue (not really a runner problem). GitHub limits run time of a CI to around 6h. If a job takes more it will be auto-cancelled (and this is what happens).

vaughnbetz, 2025-05-19:
No longer occurring ... closing.

closed 2025-05-19 by vaughnbetz