verilog-to-routing/vtr-verilog-to-routing #2172: CI check never starts

duck2, 2022-10-13:
There is a "B: Building VtR" check for merging pull requests which doesn't seem to be connected to a workflow. As a result, it always stays yellow and never lets the automatic merge to happen.

Not sure how GitHub looks up workflows from "protected branch checks", so wanted to take @hzeller's and @kgugala's opinions before just deleting that check in the settings.

Should be demonstrated in the test PR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2171

kgugala, 2022-10-18:
thanks for the ping, I'll check this

kgugala, 2022-10-18:
hmm, are you sure about this? I see the tests did start and they are being run. The one pending in the PR you linked is not run on a self-hosted runners, but on standard github runners (see: https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.github/workflows/test.yml#L67). It looks like GitHub issue?

kgugala, 2022-10-18:
I think we shoud ping @vaughnbetz about merging policy

vaughnbetz, 2022-10-18:
Tried deleting B: Building VtR and replacing it with B: Building VtR (debug,0) and B: Building VtR (release,0) checks.  (Can't see B: Building VtR in the branch protection rules dialog to add it back in).

Not sure what B: Building VtR (release, 1) does (maybe VERBOSE is on for it?).  Hence not sure if I should add it too.

[image]

duck2, 2022-10-18:
That seems to have fixed the issue. I think it's up to you to add (debug, 1) since it's pretty much the same thing as (debug, 0).

closed 2022-10-19 by duck2