verilog-to-routing/vtr-verilog-to-routing #2176: --fix_pins file.pads

Wkraipak, 2022-10-18:
HI I want to fix the IO Location and pass on the IO Location through file.pads by using the vpr option --fix_pins.
whats the format of file.pads. Is there a specific format which we have to use .
could you provide me an example for the same

Wkraipak, 2022-10-29:
Hi 
I am looking help in having the fix pins support in the vtr/vpr tool
I want the tool should use the pad file to assign the IOs
Thanks,
Wasim

vaughnbetz, 2022-11-11:
It uses the place file format, but has been renamed to --fix_clusters when you want to lock down cluster-level blocks.  See https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#cmdoption-vpr-fix_pins

The more general option is to lock down primitives though (which will implicitly constraint their clusters too): see  https://docs.verilogtorouting.org/en/latest/vpr/placement_constraints/#vpr-placement-constraints for information on that more robust and powerful option.

closed 2025-06-05 by github-actions[bot]