verilog-to-routing/vtr-verilog-to-routing #2193: post_routing_pb_pin_fixup: find_used_sink_pb_pin

amin1377, 2022-11-07:
After routing is done, "sync_netlist_to_routing" is called to accommodate the pin rotations done during the routing. One of the functions called during this operation is [find_used_sink_pb_pin](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/post_routing_pb_pin_fixup.cpp#L381). This function tries to find out the sink_pb_graph_pin by iterating over the edge's output pins. The edge's output pins are all of the pins connected to the driver_pb_pin based on the XML file, and I don't see why we expect there should be only one pin connected to driver_pb_pin!

@vaughnbetz @tangxifan

tangxifan, 2022-11-10:
@amin1377 I am on holiday this week. Will check next Thursday. If you need immediate help, please let me know.

tangxifan, 2022-11-17:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/3c1ea0885cbbeb2d42ac79d707bc232a4d59d9bf/vpr/src/pack/post_routing_pb_pin_fixup.cpp#L682-L703

@amin1377 The function [find_used_sink_pb_pin](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vpr/src/pack/post_routing_pb_pin_fixup.cpp#L381) is created for a specific need, as you see in the code comments above.

I think this function should work well in the arch "k4_N8_topology-0.85sL2-0.15gL4-on-cb-off-sb_22nm_22nm_nonLR.xml". If you see an errors, can you turn on the verbose mode: 

https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/3c1ea0885cbbeb2d42ac79d707bc232a4d59d9bf/vpr/src/base/vpr_api.cpp#L1304

If you can share the log file, I can help debugging.

closed 2024-07-05 by amin1377