verilog-to-routing/vtr-verilog-to-routing #2197: Generalize sub_tile in the grid so sub_tiles can have different width & height

vaughnbetz, 2022-11-10:
#### Proposed Behaviour
We'd like to model multi-die stack FPGAs. This is needed for the crossroads project, and will likely be useful to others.

#### Current Behaviour
The sub_tile feature almost has what we need in order to model one block under another, but it has the restriction that all sub_tiles in a tile must be the same size.  We'd like to model a large router (or large accelerator block) that is underneath multiple LABs (or maybe RAMs, etc.).  Right now we can't do this -- if we make the router large (e.g. width = 3, height = 3), then any LABs in the architecture become wide and tall.

#### Possible Solution
I think if we make the width, height, x_offset and y_offset in the grid a function of the sub_tile instead of a tile, we can have this feature.  The capacity of grid[i][j] would then give the number of sub_tiles overlapping a certain (x,y) location, and you'd go through all those sub_tiles to see where there anchor points where, what their widths and heights are etc.

#### Context
Impacting exploring routers with logic over them for the Crossroads project.

vaughnbetz, 2022-11-10:
@kmurray @tangxifan : FYI.  Sara is going to dig into this, with an eye to discussing in a future vtr meeting.

tangxifan, 2022-11-10:
@vaughnbetz Thanks for the info. Will do.

saaramahmoudi, 2022-11-21:
Different diagrams for our proposed solutions to this problem are shown below. Blocks with green are the ones that we should add to the existing code.

1. The first diagram is a straightforward approach in which we don't need to change sub-tile and tile structures or even how the grid exists. But, we need to store sub-tile attributes multiple times (as many as LAB existed on top of the NoC) 

[image]

2. The second diagram Vaughn mentioned in the previous comment as adding a new explicit z dimension.

[image]

3. The final diagram shows how we should change sub_tiles and tiles definitions and add capacity to the grid itself. 

[image]

saaramahmoudi, 2022-12-01:
@vaughnbetz 
Updating the issue after our meeting with Kevin.
Presentation is attached for @tangxifan to review and discuss later.
[Multi-die stack FPGAs.pdf](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/10128265/Multi-die.stack.FPGAs.pdf)

tangxifan, 2022-12-06:
@vaughnbetz @saaramahmoudi Thanks for the inputs with detailed explanation. 

I personally prefer the solution by adding an explicit z dimension, i.e., ``grid[i][j][z_layer]``.
Here are my reasons:

- I believe that it is clean and easy to understand for developers. Grid layout is separated for each die of the 3D stacked FPGAs.
- I suggest not to add capacity to grids, which may cause confusion between the capacity of subtiles and grids. 
- From the perspective of OpenFPGA, when generating netlists or bitstreams, ``grid[i][j][z_layer]``  allows me to perform a top-down walkthrough on FPGA fabric, which is more natural in coding. 

Actually, I am not 100% clear to which level of 3D stacked FPGAs, that VPR would like to support. My understanding is that as an architecture exploration tool, VPR aims to support very flexible FPGA architectures. I have a few questions:

- Do we allow independent/correlated routing architecture on each die? Are we going to model the routing architecture of all the dies in one RRGraph or each die may have a separated RRGraph?
- For a given tile coordinate (x, y), will all the tiles on different die have the same dimension? Also, will these tiles on different dies have different pin locations?
- If we apply ``width``, ``height``, ``x_offset`` and ``y_offset`` to subtile, it is not clear to me how to determine thhe grid/physical_tile dimension. Are we considering a sum of all the subtile dimension? Some more syntax may be required and the maths may be complicated.

These are my views based on current knowledge. I am definitely interested in the technical feature, and OpenFPGA will support it. I would like to discuss more details and open to alter my views.

vaughnbetz, 2022-12-06:
Thanks @tangxifan .  Your opinion matches the emerging consensus -- add an explicit z for clarity.  So we'll go with that.
- For routing architecture, for now we're going to leave
- the routing graph alone.  The stacked dice we're modeling in the immediate future still have only one layer of programmable routing that the lower dice connect to.  So the placement view would be generalized and the rr-graph generator would have some additional code to hook things up with some flexibility, but after that we'd have an (x,y) routing graph.  If future stacked dice have multiple programmable routing dice we could add a z dimension to the rr-graph too though.
- For a given (x,y) no restrictions will be enforced on the other z coordinates; the blocks on them could have different dimensions (and that's actually an important use case we want to explore, so this code will be exercised right away). Similarly, those tiles could have different pin outs.

tangxifan, 2022-12-06:
Thanks @vaughnbetz for the details. Now it is clear to me.
Do we expect any changes on the arch XML when supporting the 3D-stacked FPGAs?
I am thinking about where we should define the ``z`` for the grids. For example,

```
    <fixed_layout name="2x2" width="4" height="4" num_layers="2">
      <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
      <perimeter type="io" priority="100"/>
      <corners type="EMPTY" priority="101"/>
      <!--Fill with 'clb'-->
      <fill type="clb_die0" priority="10" layer="0"/>
      <fill type="clb_die1" priority="10" layer="1"/>
    </fixed_layout>
```

saaramahmoudi, 2022-12-06:
@tangxifan This is how we discuss to implement it on the architecture file. Layer tag is going to be optional, so we don't need to update existing architecture files, and die number is considered to be 0 if unspecified. We don't have any width and height attribute on the layout tag right now. Number of available layers is also can be calculated using die numbers attribute on the layer tag (will be 1 if left unspecified). Does it sound a right way to do it to you?

```
    <fixed_layout name="first_layer">
        <layer die="0">
          <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
          <perimeter type="io" priority="100"/>
          <corners type="EMPTY" priority="101"/>
          <!--Fill with 'clb'-->
          <fill type="clb_die0" priority="10"/>
        </layer>
    </fixed_layout>
    <fixed_layout name="second_layer">
        <layer die="1">
          <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
          <perimeter type="io" priority="100"/>
          <corners type="EMPTY" priority="101"/>
          <!--Fill with 'clb'-->
          <fill type="clb_die1" priority="10"/>
        </layer>
    </fixed_layout>
```

ganeshgore, 2022-12-08:
Hello, based on above discussion.
Is this the correct way to interpret how pins on different Z levels will be flattened in the 2D RRGraph?

[image]

vaughnbetz, 2022-12-08:
Discussed in meeting today. Consensus: go with z coordinate, go with Xifan's proposed syntax, make layer attribute optional /defaul to 0 so existing archs work).
For the rr-graph: current plan is not to change the rr-graph much; one layer of programmable routing still.  Devices on the z=1 layer may connect to the programmable routing via connection boxes though (or they may connect via a NoC).  If connecting to the programmable routing, their pins should have z=1 so we can tell what layer they are on.  So likely we'll wind up putting a  z-coordinate (default to 0) on the rr-graph just for annotation of the pins / drawing etc.  To minimize memory bloat, we may be able to put such a z-coordinate in the flyweight (rr-indexed data) where the cost_index points to it (since we wouldn't have many z coordinates at all).

tangxifan, 2022-12-08:
@vaughnbetz I thought twice. The shortcoming of my syntax is that it may not be easy for engineers to spot grids on specific layers, when the size of ``<layout>`` block grows. Imagine there are 100 lines of grid definition under a layout.

 I suggest to combine mine and @saaramahmoudi 's together, as follows.

```
    <fixed_layout name="2-layer-stacked-fpga" height="4" width="4">
        <layer die="0">
          <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
          <perimeter type="io" priority="100"/>
          <corners type="EMPTY" priority="101"/>
          <!--Fill with 'clb'-->
          <fill type="clb_die0" priority="10"/>
        </layer>
        <layer die="1">
          <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
          <perimeter type="io" priority="100"/>
          <corners type="EMPTY" priority="101"/>
          <!--Fill with 'clb'-->
          <fill type="clb_die1" priority="10"/>
        </layer>
    </fixed_layout>
```

This will allows us to have a clear view on how each layer look like. Engineers/developers can remove or rework a layer in a straightforward way.

It does not need to update existing architecture files. If there are no <layer> defined, we create a default layer when parsing the architecture XML.

Let me know what you think. I can help @saaramahmoudi if she needs advises on developing the parser.

vaughnbetz, 2022-12-08:
Sure, that is fine with me.

I think with your earlier syntax people could still make it readable by grouping all the layer0 grid entries together, then all the layer 1 entries, with a comment line in between if they liked.  The syntax wouldn't force them to to that, but they could if they wished.

Forcing such an organization may be cleaner though, as you suggest.  I'm OK with either.

tangxifan, 2022-12-08:
@vaughnbetz Yes. I see that people can always find a way to make their architecture XML clean. But most of time, they eventually fail to do so, due to various reasons (for instance, very tight deadlines and too many revisions). Once we have many lines in a ``<layout>`` block, it is difficult for developers to spot ``layout_id`` in each line and ensure that they are correct. Therefore, I believe we should provide developers the syntax that forces them to follow a clean way. 

In addition, it is also straightforward to implement the parser for the ``<layout>``. You can re-allocate memory by counting the number of ``<layout>`` blocks under a ``<fixed_layout>`` or a ``<auto_layout>``.  Otherwise, you may have to parse all the lines and find out the number of layers, and then allocate memory.

vaughnbetz, 2022-12-08:
Good points; I agree.

closed 2023-05-02 by saaramahmoudi