verilog-to-routing/vtr-verilog-to-routing #2226: Slow compile/link time

alaindargelas, 2023-01-13:
#### Expected Behaviour
When typing make with 0 code change, compilation should happen almost instantaneously

#### Current Behaviour
There seems to be several files that get updated every time "make" is invoked, leading to multiple compilations and linkage running for minutes. When VPR is embedded in another project's CMake (Like OpenFPGA), when compiling the overarching project for a code change (not in VPR), VPR takes the majority of the compile/link time.

#### Possible Solution

#### Steps to Reproduce

1. 
2. 

#### Context

#### Your Environment

* VTR revision used: master
* Operating System and version: Ubuntu 21.10
* Compiler version: gcc 11.2

duck2, 2023-01-14:
You can try turning off VPR's IPO (`VTR_IPO_BUILD`) for your debug builds. AFAIK that's what causes the long linking time.

alaindargelas, 2023-01-14:
It works, you are a life saver!

closed 2023-01-14 by alaindargelas