verilog-to-routing/vtr-verilog-to-routing #2229: NoC Failed to find architecture file

Ankelih, 2023-01-19:
Hello,

When I tired to run NoC with vpr I get the following error.

File: /home/deniz/vtr-verilog-to-routing/vtr_flow/benchmarks/noc/Test_Designs/test_design_for_noc_traffic_flows_feature/multiple_noc_routers.blif

The command I used is:

$VTR_ROOT/vpr/vpr $VTR_ROOT/vtr_flow/arch/noc/non_mesh_noc_topology/k6_frac_N10_frac_chain_mem32K_40nm_with_a_embedded_star_noc_topology.xml 
$VTR_ROOT/vtr_flow/benchmarks/noc/Test_Designs/test_design_for_noc_traffic_flows_feature/multiple_noc_routers.blif --route_chan_width 100

Could you please help me  to solve this problem?

Best,
Deniz

Srivat97, 2023-01-19:
Hi @Ankelih,

The reason for the failure is due to the test design not being mapped for the corresponding architecture provided. We have a more up to date set of test benchmarks and NoC enhanced FPGA architectures which I can point to.

 The most up to date NoC code can be found under the "noc_benchmark_designs_final" branch. This branch also includes a number of benchmarks which can be found in "$VTR_ROOT/vrt_flow/benchmarks/noc/Synthetic_Designs/". The architecture file that we tested the benchmarks on can be found in "$VTR_ROOT/vrt_flow/arch/noc/mesh_noc_topology/stratixiv_arch.timing_with_a_embedded_10X10_mesh_noc_topology.xml".

Below would be an example run:
"$VTR_ROOT/vpr/vpr $VTR_ROOT/vrt_flow/arch/noc/mesh_noc_topology/stratixiv_arch.timing_with_a_embedded_10X10_mesh_noc_topology.xml $VTR_ROOT/vrt_flow/benchmarks/noc/Synthetic_Designs/simple_2_noc_1D_chain/simple_2_noc_1D_chain.blif --device EP4SE820 --route_chan_width 300 --noc on --noc_flows_file $VTR_ROOT/vrt_flow/benchmarks/noc/Synthetic_Designs/simple_2_noc_1D_chain/simple_2_noc_1D_chain.flows --noc_routing_algorithm xy_routing"

The command above will invoke NoC related features within VPR. For other designs just change the circuit file location and the traffic flows file location. Each benchmark has an associated traffic flows file which describe the communication between routers in the NoC  (they can be found in the same location as all the benchmark circuits). By default VPR is BFS for routing within the NoC and will be used in the "--noc_routing_algorithm" option is omitted.

Ankelih, 2023-02-03:
Hi @Srivat97,

Sorry I couldn't reply back. I successfully run the flow with the command you provide. However after that I tried to generate a blif file with very much same verilog codes that already exist. While using ODIN, the tool eliminates the first router. I guess since the output ports of the first router are open it ingores the first router. How do you generate blif files for the noc architecture.

Best

Srivat97, 2023-02-07:
Hi @Ankelih,

You could generate the blif files with ODIN2 but have the output ports connected to "filler' signals. We have another flow called Titan which uses Quartus as the front end synthesis tool (currently we target stratixiv based architectures). The documentation and setup for this flow will be available in a few weeks. You should be able to generate blif files for existing verilog code using this flow. I will send another reply to this thread when this becomes available.

closed 2025-06-06 by github-actions[bot]