verilog-to-routing/vtr-verilog-to-routing #2242: The congestion heat map is invisiable

asdddddasfwe, 2023-02-14:
#### My Behavior
As it is said in document, I used the --disp on commend to open the graphic page. And than I use the Congestion button to see the congestion heat map.

#### What I have saw
I can't see the congestion heat map which is beyond my expectation. It just appear a blue bar near the original image. I don't know how to fix it.

#### Your Environment

* VTR revision used:The master version on github
* Operating System and version:Ubuntu 22.10
* Compiler version:

nkunkunku, 2023-02-17:
I am a VTR user, can I ask you a question?
I use the "--disp on" command provided in the document but no graphical interface for the layout pops up. The command I use is "/vpr$ ./vpr EArch.xml alu4.blif --place --disp on ", Do you know how to solve it? Does this system version matter?

Another question is, can I compile this project under windows system?

asdddddasfwe, 2023-02-17:
Do you use the vtr 8.0.0?

asdddddasfwe, 2023-02-17:
The vtr 8.0.0 can find the libx11 even I have install it. So it's graphics function is disable. I suggest you to install the master version.

closed 2025-06-06 by github-actions[bot]