verilog-to-routing/vtr-verilog-to-routing #2247: 'tileable' found on node 'layout'

zliu1Charlotte, 2023-02-22:
this is the command I ran:
`~/vtr-verilog-to-routing/vpr/vpr k4_N4_tileable_130nm.xml xor_cipher.blif --clock_modeling route`

it gives error:
```
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+f669015f3
Revision: v8.0.0-6941-gf669015f3
Compiled: 2023-02-10T16:13:42
Compiler: GNU 9.4.0 on Linux-5.15.0-58-generic x86_64
Build Info: release IPO VTR_ASSERT_LEVEL=2

University of Toronto
verilogtorouting.org
vtr-users@googlegroups.com
This is free open source code under MIT license.

VPR was run with the following command-line:
/home/pc-1/vtr-verilog-to-routing/vpr/vpr k4_N4_tileable_130nm.xml xor_cipher.blif --clock_modeling route

Architecture file: k4_N4_tileable_130nm.xml
Circuit name: xor_cipher

# Loading Architecture Description
Warning 1: Model 'io' input port 'outpad' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 2: Model 'io' output port 'inpad' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
# Loading Architecture Description took 0.00 seconds (max_rss 16.3 MiB, delta_rss +1.5 MiB)
Error 1: k4_N4_tileable_130nm.xml:65 Unexpected attribute 'tileable' found on node 'layout'.
The entire flow of VPR took 0.00 seconds (max_rss 16.3 MiB)
```

Martoni, 2024-09-15:
Got the same problem with [clear](https://github.com/efabless/clear?tab=readme-ov-file#steps-for-using-openfpga-to-generate-bitstream-for-sofa-fpga-fabric) and [vpr_arch.xml](https://github.com/efabless/OpenFPGA_bitstream_generation/blob/SOFA/openfpga_flow/tasks/SOFA_tasks/arch/vpr_arch.xml) : 

```shell
$ vpr vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH/arch/vpr_arch.xml vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH/counter.blif
VPR FPGA Placement and Routing.
Version: 8.1.0-dev+v8.0.0-10995-gf13f87b5a-dirty
Revision: v8.0.0-10995-gf13f87b5a-dirty
Compiled: 2024-09-11T22:17:57
Compiler: GNU 11.4.0 on Linux-5.15.0-89-generic x86_64
Build Info: release IPO VTR_ASSERT_LEVEL=2

University of Toronto
verilogtorouting.org
vtr-users@googlegroups.com
This is free open source code under MIT license.

VPR was run with the following command-line:
vpr vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH/arch/vpr_arch.xml vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH/counter.blif

Architecture file: vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH/arch/vpr_arch.xml
Circuit name: counter

# Loading Architecture Description
Warning 1: Model 'io' input port 'outpad' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 2: Model 'io' output port 'inpad' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 3: Model 'frac_lut4' input port 'in' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 4: Model 'frac_lut4' output port 'lut4_out' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 5: Model 'frac_lut4' output port 'lut3_out' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 6: Model 'frac_lut4' output port 'lut2_out' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
Warning 7: Model 'carry_follower' input port 'cin' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 8: Model 'carry_follower' input port 'b' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 9: Model 'carry_follower' input port 'a' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input)
Warning 10: Model 'carry_follower' output port 'cout' has no timing specification (no clock specified to create a sequential output port, not combinationally connected to any inputs, not a clock output)
# Loading Architecture Description took 0.00 seconds (max_rss 16.1 MiB, delta_rss +1.2 MiB)
Error 1: vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH/arch/vpr_arch.xml:156 Unexpected attribute 'tileable' found on node 'layout'.
The entire flow of VPR took 0.00 seconds (max_rss 16.1 MiB)
```

Got this in line 156 and neighborhood  : 

```xml
  <!-- ODIN II specific config ends -->
  <!-- Physical descriptions begin -->
  <layout tileable="true">
    <auto_layout aspect_ratio="1.0">
      <!--Perimeter of 'io' blocks with 'EMPTY' blocks at corners-->
      <row type="io_top" starty="H-1" priority="100"/>
      <row type="io_bottom" starty="0" priority="100"/>
      <col type="io_left" startx="0" priority="100"/>
      <col type="io_right" startx="W-1" priority="100"/>
      <corners type="EMPTY" priority="101"/>
      <!--Fill with 'clb'-->
```

Did you resolve this problem ?

ganeshgore, 2024-09-15:
@Martoni the tileable feature is only supported in OpenFPGA for now. The effort for merging that to VPR main is still ongoing. Meanwhile, you compile the VPR on the [openfpga](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/openfpga) branch to get this functionality.

closed 2025-10-03 by github-actions[bot]