verilog-to-routing/vtr-verilog-to-routing #2248: VPR failed when running with the architecture description file that comes with VTR

jdzhu19, 2023-02-23:
#### Expected Behaviour
VPR should be successful

#### Current Behaviour
When I use the arch in [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/COFFE_22nm/stratix10_arch.xml](url) and benchmark in [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/doc/src/quickstart/blink.v](url) as the input to run vtr by instruction:
```
$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py \
$VTR_ROOT/doc/src/quickstart/blink.v \
$VTR_ROOT/vtr_flow/arch/COFFE_22nm/stratix10_arch.xml \
```
, VPR failed because (error in vpr.out)
```
Failed to route connection from 'blink^ADD~2-0[0]' to 'out:blink^o_led' for net 'blink^o_led' (#1)
Routing failed for net 1
```
When I use the arch in [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/xilinx/simple-7series.xml](url) and benchmark in [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/doc/src/quickstart/blink.v](url)  as the input to run vtr by instruction:
```
$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py \
$VTR_ROOT/doc/src/quickstart/blink.v \
$VTR_ROOT/vtr_flow/arch/xilinx/simple-7series.xml \
```
, VPR failed because (error in vpr.out)
```
Failed to route connection from 'n27' to 'n27' for net 'blink^r_counter~1_FF' (#2)
Routing failed for net 2
```
When I use the arch in [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/xilinx/simple-7series.xml](url) and benchmark in [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/benchmarks/verilog/diffeq1.v](url)  as the input to run vtr by instruction:
```
$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py \
$VTR_ROOT/vtr_flow/benchmarks/verilog/diffeq1.v \
$VTR_ROOT/vtr_flow/arch/xilinx/simple-7series.xml \
```
A large number of `Failed route at end, repack cluster trying detailed routing at each stage.` occur in vpr.out

But when I use [https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/arch/timing/EArch.xml](url) and various benchmarks to run vtr, everything is fine.

#### Possible Solution

#### Steps to Reproduce
Use the arch files and design files linked above.

#### Context
I want to use COFFE to export the architecture description file for input into VTR, but I get failures and errors when testing the architecture description file that comes with VTR.

#### Your Environment

* VTR revision used: Latest master branch
* Operating System and version:
* Compiler version:

luck-codeer, 2023-07-21:
May I ask if you have resolved this issue? I have also encountered a similar problem.

vaughnbetz, 2023-07-25:
@MohamedEldafrawy : this is an architecture from your MASc.  Any ideas?

jdzhu19, 2023-07-25:
> May I ask if you have resolved this issue? I have also encountered a similar problem.

To be honest, this is a problem I just encountered in this project. It has been so long that I forgot how to solve it at that time, but what is certain is that this combination can run through the CAD flow.

vaughnbetz, 2023-07-25:
@StephenMoreOSU : not sure if you remember the fix.  It would be good to integrate ...

StephenMoreOSU, 2023-07-28:
@tangxifan This looks to be an error in the post routing netlist / clustering fixup. Can you or someone on your team take a look? Thanks :)

StephenMoreOSU, 2023-07-28:
@kimiatkh I heard you may have hit this issue as well

luck-codeer, 2023-10-31:
When encountering such an issue, one might consider using the command "../scripts/run_vtr_task.py power/power_nofrac_22nm" as an alternative solution.

OlinLai, 2023-12-12:
> When encountering such an issue, one might consider using the command "../scripts/run_vtr_task.py power/power_nofrac_22nm" as an alternative solution.

Hi, I also met with problem of stratix10_arch.xml but I am a little bit confused about your alternative solution.
Running this command "../scripts/run_vtr_task.py power/power_nofrac_22nm" means using another architecture.
[image]

So are you saying that when we want to use stratix10_arch.xml, we need to use run_vtr_task.py instead of run_vtr_flow.py? Because when I use run_vtr_task.py, I still encounter the same routing failure as with run_vtr_flow.py.

Also, I would like to ask, does run_vtr_task.py use Yosys or Odin+ABC as the synthesis tool?

Thanks for your help!

luck-codeer, 2023-12-13:
Well, what I meant is to conduct the experiment using .xml in run_task.py, not with stratix10_arch.xml. Also, I believe it will use the corresponding synthesis tool. Here are some of the results of its execution. I hope it will help you.
[image]

closed 2025-06-06 by github-actions[bot]

vaughnbetz, 2025-06-06:
@duck2 made changes to generalize the post routing fix up so this may be fixed. In any case without someone driving it to test it I'm going to let it close.