verilog-to-routing/vtr-verilog-to-routing #2257: Need to migrate off Ubuntu 18.04

Tulong4Dev, 2023-03-16:
#### Proposed Behaviour
- Workflows using the ubuntu-18.04 YAML workflow label should be updated to ubuntu-20.04, ubuntu-22.04, or ubuntu-latest

#### Current Behaviour
- Workflows using the ubuntu-18.04

#### Possible Solution
- Update YAML workflow label to ubuntu-20.04, ubuntu-22.04, or ubuntu-latest

#### Context

- The Ubuntu 18.04 Actions runner image is being deprecated and will be removed by 4/1/23
- Workflows using the ubuntu-18.04 YAML workflow label should be updated to ubuntu-20.04, ubuntu-22.04, or ubuntu-latest

vaughnbetz, 2023-03-16:
I suggest Ubuntu 22.04 so we have 4 years before we have to do this again, but we're on a more stable (LTS) version.

tangxifan, 2023-04-05:
Close as major problem as been resolved in #2258
We see some CI failures to fix up. Will create follow-up issues to address each of them.

closed 2023-04-05 by tangxifan

tangxifan, 2023-04-05:
Three follow-up issues created:

#2268 
#2269 
#2270