verilog-to-routing/vtr-verilog-to-routing #2262: List of things to add/update regarding NoC placement

Srivat97, 2023-03-25:
- [ ] Remove check_recomputed_noc_latency_cost() function and replace with a single comparison to MIN_EXPECTED_NOC_LATENCY_COST
- [ ] Update propose_router_swap() function to not iterate through all router cluster blocks to choose a random block to swap. This is currently O(n), improve run time of this function
- [ ] Group all instance of noc_latency_cost and noc_aggregate_bandwidth cost together within a struct
- [ ] Look to using VTR strong IDs (NocTrafficFlowId) directly in loop counters rather than casting (see if this is possible). Look at all placer we currently cast in NoC related code
- [ ] Change ::increment and ::decrement enums to +1 and -1 and then multiply the value the change in bandwidth by this value and adding to the links current bandwidth
- [ ] Look at changing params of pointers to params of references in all NoC related functions
- [ ] Look to see whether we can store traffic flow indices in associated_traffic_flows vector (this is related to noc_traffic_flows_storage datastructure and how we currently store associated traffic flows to a rotuer block). Then we can compare to NoCFlowID::Invalid() rather than null pointer
- [ ] Look at making try_swap and try_place smaller by grouping NoC related code into functions and adding them to noc_place_utils

vaughnbetz, 2023-05-12:
@saaramahmoudi : please check off items as they are done.

vaughnbetz, 2023-05-12:
All items done except 

- [ ] Iterating over strongIDs (needs ++ which doesn't exist). In most cases (all cases?) we should be able to use a range-based for to iterate over them.  Should do that where we can, and identify if there are places where we can't do this.  That could be motivation for adding ++ and -- to StringIDs, but I'd ask @kmurray first as it's likely he knows of an alternative (since he made the stringIDs and didn't include ++ and --).
- Not able to make try_swap and try_place smaller with more NoC helper functions.

vaughnbetz, 2023-05-12:
Looks like we can get the entire pair (key,value) from a map with a range-based for, so doesn't seem like we need ++ and -- on VTRStrongID.

vaughnbetz, 2023-05-12:
So only one thing left: try to get rid of the casts and then this can be merged.

closed 2023-05-19 by saaramahmoudi