verilog-to-routing/vtr-verilog-to-routing #2266: Wrong blif for subtractor for different architecture files

FPGA1234, 2023-03-30:
We have tried to execute the following RTL for subtracting two 4-bit numbers using the architecture file which supports carry chain adder. But the blif generated by the tools is giving the errors.

RTL code is shown below -
[image]

The tools is generating a blif file with the following logic -
[image]

The behavior of the tool output vs expected behavior is elaborated is shown below -
[image]

In the image, one could see that subtractor implemented by tool has the adder (highlighted in red circle) with inputs as GND and VCC, CIN as GND, thus giving COUT as GND.
This results in computation of 2's complement as 1's complement instead of (1's complement+1) as 2's complement = (1's complement + 1).
Also, one could see in the blif that the sign bit of output out, i.e., out[4] is assigned value of 0.

We have analyzed and found the working solution as shown in the diagram on the right.
The changes are highlighted in green and blue circle.
Now, we have adder (highlighted in green circle) with inputs as VCC and VCC, CIN as GND, thus giving COUT as VCC to make 2's complement = (1's complement + 1).
Also, we have taken an additional adder (highlighted in blue circle) with inputs as GND and VCC, CIN as COUT of previous adder. Thus, generating correct sign bit.

The architecture file is added below -
[k6_frac_N10_adder_chain_40nm.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/11108686/k6_frac_N10_adder_chain_40nm.txt)

One can reproduce the error by giving by calling vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_flow.py using the architecture file, RTL provided above.

alirezazd, 2023-03-30:
@vaughnbetz Please assign me to this issue.

vaughnbetz, 2025-05-22:
Sending to Logan to see if he can test if this is still an issue with yosys+parmys.

vaughnbetz, 2025-05-22:
@loglav03

loglav03, 2025-06-16:
@vaughnbetz - It looks like this is still an issue. The initial inputs a and b are showing as GND and VCC instead of both being VCC, and out~4 is showing 0.

[image]

loglav03, 2025-06-17:
I believe this error is in parmys. Alireza did some work on this in the past but was unable to finish. I'm going to to pick up where he left off and see if I get anywheres with it.