verilog-to-routing/vtr-verilog-to-routing #2267: Problems in making the latest VTR

jdzhu19, 2023-04-05:
#### Expected Behaviour
The make should be completed smoothly.

#### Current Behaviour
When I use the pre-February version of VTR, make goes fine and works fine, but when I use the post-February version, make reports the following error, and since the February version seems to have updated parmys and deprecated ODIN, I suspect it's a problem in that regard.
```
[ 23%] Building C object abc/CMakeFiles/libabc.dir/src/base/acb/acbAig.c.o
passes/sat/formalff.cc:85:13: error: call to member function 'get_drivers' is ambiguous                                                                                                            modwalker.get_drivers(portbits, {bit});
                ~~~~~~~~~~^~~~~~~~~~~
./kernel/modtools.h:412:14: note: candidate function
        inline bool get_drivers(pool<PortBit> &result, RTLIL::SigBit bit) const
                    ^
./kernel/modtools.h:500:7: note: candidate function
        bool get_drivers(pool<PortBit> &result, RTLIL::SigSpec signal) const
             ^
./kernel/modtools.h:453:14: note: candidate template ignored: couldn't infer template argument 'T'
        inline bool get_drivers(pool<PortBit> &result, const T &bits) const
                    ^
passes/sat/formalff.cc:221:13: error: call to member function 'get_consumers' is ambiguous
                modwalker.get_consumers(portbits, {bit});
                ~~~~~~~~~~^~~~~~~~~~~~~
./kernel/modtools.h:423:14: note: candidate function
        inline bool get_consumers(pool<PortBit> &result, RTLIL::SigBit bit) const
                    ^
./kernel/modtools.h:506:7: note: candidate function
        bool get_consumers(pool<PortBit> &result, RTLIL::SigSpec signal) const
             ^
./kernel/modtools.h:466:14: note: candidate template ignored: couldn't infer template argument 'T'
        inline bool get_consumers(pool<PortBit> &result, const T &bits) const
                    ^
[ 23%] Building C object abc/CMakeFiles/libabc.dir/src/base/acb/acbCom.c.o
......
2 errors generated.
make[4]: *** [passes/sat/formalff.o] Error 1
make[4]: *** Waiting for unfinished jobs....
```

#### Steps to Reproduce
I used version 99d71d9 which was committed on Mar.25, because it was the latest version I found that passed all the checks.
So all you need to do is to clone it and make it.

#### Your Environment

* VTR revision used: commit 99d71d9

poname, 2023-04-07:
Did you run the `install_apt_packages.sh` script before `make`?

jdzhu19, 2023-04-07:
Oh, I didn't pay too much attention to install_apt_packages.sh because I was able to make VTR and Yosys on a recent version (commit f669015on Jan 24), and my system is centos, so I can't install directly with apt-get command. I'll check it later, thank you!

jdzhu19, 2023-04-14:
> Did you run the `install_apt_packages.sh` script before `make`?

Hello! I have run the install_apt_packages.sh script before make, but it seems that there is a similar error to the previous one that still exists.
`[  6%] Building CXX object libs/librtlnumber/CMakeFiles/rtl_number.dir/main.cpp.o
passes/sat/formalff.cc:85:13: error: call to member function 'get_drivers' is ambiguous                                                                                                            modwalker.get_drivers(portbits, {bit});
                ~~~~~~~~~~^~~~~~~~~~~
./kernel/modtools.h:412:14: note: candidate function                                                                                                                                       inline bool get_drivers(pool<PortBit> &result, RTLIL::SigBit bit) const
                    ^
./kernel/modtools.h:500:7: note: candidate function                                                                                                                                        bool get_drivers(pool<PortBit> &result, RTLIL::SigSpec signal) const
             ^
./kernel/modtools.h:453:14: note: candidate template ignored: couldn't infer template argument 'T'                                                                                         inline bool get_drivers(pool<PortBit> &result, const T &bits) const
                    ^
passes/sat/formalff.cc:221:13: error: call to member function 'get_consumers' is ambiguous
                modwalker.get_consumers(portbits, {bit});
                ~~~~~~~~~~^~~~~~~~~~~~~
./kernel/modtools.h:423:14: note: candidate function
        inline bool get_consumers(pool<PortBit> &result, RTLIL::SigBit bit) const
                    ^
./kernel/modtools.h:506:7: note: candidate function
        bool get_consumers(pool<PortBit> &result, RTLIL::SigSpec signal) const
             ^
./kernel/modtools.h:466:14: note: candidate template ignored: couldn't infer template argument 'T'
        inline bool get_consumers(pool<PortBit> &result, const T &bits) const
                    ^
[  6%] Building CXX object libs/libvtrutil/CMakeFiles/libvtrutil.dir/src/vtr_list.cpp.o`

poname, 2023-04-14:
> Hello! I have run the install_apt_packages.sh script before make, but it seems that there is a similar error to the previous one that still exists. `[ 6%] Building CXX object [...]

Please provide more details on the operating system and the compiler tool chain you are using.
Our CI testing is on Ubuntu 22.04, so that is the best tested platform and recommended for development.

jdzhu19, 2023-04-14:
> 

I'm using centos 7 and I've seen mention of Ubuntu being the most compatible system, but our servers are almost all on that system. And the version of gcc used is 12.2. 
Also, I would like to ask, is it possible to compile parmys with YOSYS_PARMYS_PLUGIN for previous commit versions, for example, [f669015f3c](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/f669015f3c7adf4347b5cb6e7b4013ef0c9c31bb), I don't seem to see it mentioned in the yosys related docs that come with this commit version, but it is mentioned in CMakeLists.

jdzhu19, 2023-04-14:
> Gcc 12 is supported and the related CI task is green for the latest VTR version [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4695156247/jobs/8324046748). It seems to be a local problem, rather than a VTR build issue.
> 
> Please check the outdated `run` documentation [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/f669015f3c7adf4347b5cb6e7b4013ef0c9c31bb/doc/src/vtr/run_vtr_flow.rst):
> 
> ```
>     # Using the Parmys (Partial Mapper for Yosys) plugin as partial mapper
>     ./run_vtr_flow <path/to/Verilog/File> <path/to/arch/file> -start yosys -mapper parmys
> ```
> 
> Will run the VTR flow (default configuration) with Yosys frontend using Parmys plugin as partial mapper. To utilize the Parmys plugin, the `-DYOSYS_PARMYS_PLUGIN=ON` compile flag should be passed while building the VTR project with Yosys as a frontend. run `make CMAKE_PARAMS="-DWITH_YOSYS=ON -DYOSYS_PARMYS_PLUGIN=ON"` for the old version mentioned above.
> 
> The latest VTR is utilizing the Parmys frontend by default and no extra configuration is needed.

Copy that, thanks for the quick reply, I'll do some more error checking.

jdzhu19, 2023-04-14:
> Gcc 12 is supported and the related CI task is green for the latest VTR version [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4695156247/jobs/8324046748). It seems [...]

Sorry to bother again, in the old version I mentioned above I installed parmys using `make CMAKE_PARAMS="-DWITH_YOSYS=ON -DYOSYS_PARMYS_PLUGIN=ON",` the make process went well and when I executed with -start yosys it produced metrics successfully, but when I run -start yosys -mapper parmys I get the following error 
```
ERROR: Can't load module `. /parmys': 
/home/jzhu484/DSE/VTR/Yosys/bin/... /share/yosys/plugins/parmys.so: undefined symbol: _ZN5Yosys4PassC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES6_
```
My guess is that the corresponding downloaded parmys version does not match the yosys version, maybe the auto-downloaded parmys version is too new? I would like to ask your opinion and if this is the case is there any solution?
Thank you for your time again, I look forward to your response.

poname, 2023-04-14:
Here the version is specified for the commit mentioned above:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/f669015f3c7adf4347b5cb6e7b4013ef0c9c31bb/libs/EXTERNAL/CMakeLists.txt#L148-L149

So the issue is not related to the Parmys version.

You may try `rm -rf build` to remove any cached build files and re-build the project, it may work, otherwise it is hard to tell since there are other factors.

jdzhu19, 2023-04-15:
> Here the version is specified for the commit mentioned above: [...]

Well, that still doesn't work, in addition to the operations you mentioned, I also re-clone the VTR project and then compile it with `make CMAKE_PARAMS="-DWITH_YOSYS=ON -DYOSYS_PARMYS_PLUGIN=ON",` but I still get the same error every time I do `-start yosys -mapper parmys` in the task, I'll try something else, thanks.

poname, 2023-04-28:
@jdzhu19 I found some time to test centos, the centos7 is way too old to get dependencies right, so I tried centos8 and it worked fine. Created a centos8 container and install the dependencies, built and ran vtr!
The chain of commands from history:
```
   17  dnf install gcc-toolset-11-gcc gcc-toolset-11-gcc-c++
   21  source /opt/rh/gcc-toolset-11/enable
   22  gcc --version
   23  g++ --version
   24  yum install gcc openssl-devel bzip2-devel libffi-devel zlib-devel -y
   25  dnf install bzip2-devel expat-devel gdbm-devel     ncurses-devel openssl-devel readline-devel wget     sqlite-devel tk-devel xz-devel zlib-devel libffi-devel
   27  cd /home/
   31  wget https://www.python.org/ftp/python/3.8.1/Python-3.8.1.tgz
   32  tar -xf Python-3.8.1.tgz 
   33  cd Python-3.8.1
   35  ./configure --enable-optimizations
   36  make -j10
   37  make altinstall
   38  cd ..
   39  dnf install cmake
   41  dnf install -y     make     cmake          pkg-config     bison     flex 
   44  dnf install clang
   45  dnf install bison
   48  dnf install readline-devel
   49  dnf install gawk
   51  dnf install tcl-devel
   54  dnf install zlib-devel
   55  dnf install git
   56  git clone https://github.com/verilog-to-routing/vtr-verilog-to-routing.git
   57  cd vtr-verilog-to-routing/
   58  make env
   65  ln -s /usr/local/bin/python3.8 /usr/bin/python
   66  ln -s /usr/local/bin/python3.8 /usr/bin/python3
   69  ln -s /usr/local/bin/pip3.8 /usr/bin/pip
   70  ln -s /usr/local/bin/pip3.8 /usr/bin/pip3
   71  make env
   72  source .venv/bin/activate
   73  pip install -r requirements.txt
   74  make -j10
   80  dnf install -y time
   76  ./run_reg_test.py parmys_reg_basic -j10
```
Attached are screenshots as POC.

[image]
[image]

```
(.venv) [root@adb803c75f53 vtr-verilog-to-routing]# gcc --version
gcc (GCC) 11.2.1 20210728 (Red Hat 11.2.1-1)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(.venv) [root@adb803c75f53 vtr-verilog-to-routing]# clang --version
clang version 12.0.1 (Red Hat 12.0.1-4.module_el8.5.0+1025+93159d6c)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
(.venv) [root@adb803c75f53 vtr-verilog-to-routing]# python --version
Python 3.8.1
(.venv) [root@adb803c75f53 vtr-verilog-to-routing]# cmake --version
cmake version 3.20.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(.venv) [root@adb803c75f53 vtr-verilog-to-routing]#

(.venv) [root@adb803c75f53 vtr-verilog-to-routing]# cat /etc/os-release
NAME="CentOS Linux"
VERSION="8"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Linux 8"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:8"
HOME_URL="https://centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"
CENTOS_MANTISBT_PROJECT="CentOS-8"
CENTOS_MANTISBT_PROJECT_VERSION="8"
```

 I would recommend closing this issue.

jdzhu19, 2023-04-29:
OK, thank you so much for your time!

closed 2023-04-29 by jdzhu19