verilog-to-routing/vtr-verilog-to-routing #2268: Clang format check failures due to an upgrade in Clang-format version

tangxifan, 2023-04-05:
#### Expected Behaviour
The code format check is essential to enforce coding convention, as well as to make sure that code review focus on necessary changes only.

#### Current Behaviour

However, due to a must-have upgrade in clang-format version (PR #2258), the code format check failed.
The root of the cause is that even using the configuration file [.clang-format](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/.clang-format), the ``clang-format-14`` (default clang version on Ubuntu 22.04 LTS) apply code formatting differently than the previous version ``clang-format-7``.
Detailed error log can be found at 
[5_Test.txt](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/11161745/5_Test.txt)

#### Possible Solution

There are two ways to resolve the issue:

- Update codes w.r.t. the latest clang-format-14 results. This is straightforward but may impact some PRs which are modifying the codes which are conflicted with format results.
- Update .clang-format, which could be time consuming but have less impact on other PRs.

#### Steps to Reproduce

1. Git clone on the current master
2. Run code format by 

```
./dev/check-format.sh
```

#### Context

#### Your Environment

* VTR revision used: [ab2e17f](https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/ab2e17f07f5ccd33efdf18c0ef46c6c2f91c6391)
* Operating System and version: Ubuntu 22.04 LTS
* Compiler version: Clang-format-14

tangxifan, 2023-04-05:
@vaughnbetz I have summarized the code format errors seen in #2258 
There are two ways to solve it (see details in the issue). Waiting for your suggestion.

vaughnbetz, 2023-04-05:
Let's do the simplest thing: update the code format by running the new make format.  If there are any PRs that need minor updates / have conflicts due to that it is OK.  I think that will be faster and more maintainable in the end.

tangxifan, 2023-04-06:
@vaughnbetz No problem. We will follow this way.

@Tulong4Dev Would you mind to create a PR to clear all the formatting issues? Thanks!

closed 2023-04-06 by vaughnbetz