verilog-to-routing/vtr-verilog-to-routing #2271: Executable yosys failed

saaramahmoudi, 2023-04-05:
After [PR #2234](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2234) merged a while ago, we couldn't build master code without the error on our side. Then, the make error went away (I assume Daniel fixed it after Thursday meeting). However, we can make it now but we run any circuit that requires yosys, we get "Executable yosys failed". 

#### Expected Behaviour

yosys should work fine when we run VTR benchmarks.

#### Current Behaviour

Executable yosys failed.

#### Steps to Reproduce

1. Clone the master branch and make it.
2. Try to run VTR benchmarks.

#### Context

I am trying to land a branch into master branch but I am failing some QoR tests which I can't examine since I can't run yosys on my local machine. @MohamedElgammal and @amin1377 have a same issue on their local end. 
In addition to that, Valgrind memory check on [my branch](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4567587699/jobs/8061568376) is also failing due to "yosys executable failed" error message as well.

#### Your Environment

* VTR revision used: VTR master branch
* Operating System and version: Ubuntu - 18.04

saaramahmoudi, 2023-04-05:
@vaughnbetz Can you please mention Daniel under this issue? I can't find his Github username. Thanks!

vaughnbetz, 2023-04-06:
@alirezazd : can you or daniel take a look?  I'm not sure of Daniel's github id either.

alirezazd, 2023-04-06:
@vaughnbetz I will mention Daniel here, he will take a look a this. Thanks for posting the issue. 
@poname could you please check this? Thank you!

poname, 2023-04-06:
@vaughnbetz I'll check and get back to you soon. My first guess would be the Yosys dependencies.

poname, 2023-04-06:
@saaramahmoudi Could you install the following dependencies and give it a try
```
sudo apt-get install build-essential clang bison flex \
	libreadline-dev gawk tcl-dev libffi-dev git \
	graphviz xdot pkg-config python3 libboost-system-dev \
	libboost-python-dev libboost-filesystem-dev zlib1g-dev
```

saaramahmoudi, 2023-04-06:
@poname Thanks for your help. I installed these dependencies and I can run yosys on my local computer now.

vaughnbetz, 2023-04-06:
Great.  It seems we need a dependency install script update and maybe a documentation update (unless an existing script that's documented will do everything).  Is that part of that PR?

jdzhu19, 2023-04-07:
> We already have the documentation on the dependencies here: [prerequisites](https://docs.verilogtorouting.org/en/latest/parmys/quickstart/#prerequisites)
> 
> And I believe an update to the dependency installer script is not critical, since there were no problems with the Github CI on runners. Maybe, the developers forgot to run the `install_apt_packages.sh` after cloning. Anyway, I have updated the script to include optional Yosys dependencies as well to prevent any potential problems in the future.
> 
> **But, running the `install_apt_packages.sh` script is always required as before (once after cloning for the first time).**

Sorry to interrupt, but could you take a look at question #2267, please?

vaughnbetz, 2023-04-07:
1. A bunch of the dependencies listed above:
gawk libffi-dev git graphviz xdot libboost-system-dev libboost-python-dev libboost-filesystem-dev zlib1g-dev

are not in the install_apt_packages.sh script.  Some are probably not needed, but from https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/2267 it seems likely that some are. @poname and @alirezazd  can you run some tests to figure out what dependencies are truly needed (maybe it's all those missing ones0 and make sure they are added to install_apt_packages.sh ? 

2. The dependencies list at https://docs.verilogtorouting.org/en/latest/parmys/quickstart/#prerequisites doesn't exactly match the install_apt_packages.sh either.  It lists:     
   python3-dev \
    python3-venv
instead of python3.  It think this should be updated to list the dev versions.

3. We no longer support gcc 5.x.  We should update the dependencies at https://docs.verilogtorouting.org/en/latest/parmys/quickstart/#prerequisites to say g++ 9.x
We should also say Ubuntu 22.04 is the primary platform tested.

vaughnbetz, 2023-04-07:
Should have read all my email first :).
Thanks for fixing item 1 above already @poname . #2273  fixes item 1.  But we should still do number 2 to 4 above -- can you also take care of them @poname ?

poname, 2023-04-09:
@vaughnbetz Items 2 to 4 are done in [PR 2279](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2279). Ready to merge.

saaramahmoudi, 2023-04-17:
@poname 
Just a follow-up question; Me and @MohamedElgammal  tried all the above instruction on our group server and still getting an error while Making the master branch.

Here is the instruction we followed:
1. Cloned the master branch
2. Ran "install_apt_packages.sh"
3. Installed Python Packages mentioned in the "requirement.txt" except for Black which we already have a higher version installed. 
4. Used Make command to build.

The error we are getting is: "CMake Error: The source directory "/home/elgamma8/research/vtr_master/vtr-verilog-to-routing/build/parmys" does not appear to contain CMakeLists.txt"

Although the instruction works fine in our local machines. Do you have any idea what might be our issue?

poname, 2023-04-17:
@saaramahmoudi Can you provide more details on the operating system, the compiler tool chain and the exact command you are using for making the project.

MohamedElgammal, 2023-04-17:
OS: "Ubuntu 18.04.4 LTS"
compiler: "gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0"
command: "make BUILD_TYPE=release -j20"

poname, 2023-04-18:
I believe Vaughn mentioned [here](https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/2271#issuecomment-1500456179) that `gxx 9.x` should be the minimum standard tool chain supported and `Ubuntu 22.04` is the primary platform tested, however, [this](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4174512383/jobs/7228181359?pr=2234) CI check was green for `Ubuntu 18.04` and `gcc 7` on Feb 23. So maybe something in between has been changed that breaks support for `gcc 7` or the problem is local on the machine. I suggest to upgrade the compiler tool chain. @MohamedElgammal @saaramahmoudi

saaramahmoudi, 2023-04-20:
@poname We upgraded the compiler tool chain to 9.4.0 but still the same error exists. 

I tried on another machine with the following details and got the same error. 
OS: "Ubuntu 16.04.7 LTS"
compiler: "g++ (Ubuntu 9.4.0-1ubuntu1~16.04) 9.4.0"
command: "make BUILD_TYPE=release -j20"

The error is:
**"CMake Error: The source directory "/home/mahmo494/RL_experiment/vtr-verilog-to-routing/build/parmys" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
parmys/CMakeFiles/parmys-plugin.dir/build.make:57: recipe for target 'parmys/CMakeFiles/parmys-plugin' failed
make[3]: *** [parmys/CMakeFiles/parmys-plugin] Error 1
CMakeFiles/Makefile2:2731: recipe for target 'parmys/CMakeFiles/parmys-plugin.dir/all' failed
make[2]: *** [parmys/CMakeFiles/parmys-plugin.dir/all] Error 2"**

poname, 2023-04-20:
@saaramahmoudi Thanks for the update.