verilog-to-routing/vtr-verilog-to-routing #2294: Fix compiler warnings in vqm2blif

vaughnbetz, 2023-04-25:
#### Expected Behaviour
We should have no compiler warnings.

#### Current Behaviour
Some warnings in vqm2blif:

[ 50%] [BISON][VqmParser] Building parser with bison 3.8.2
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_parser.y:45.1-14: warning: deprecated directive: ‘%error-verbose’, use ‘%define parse.error verbose’ [-Wdeprecated]
   45 | %error-verbose
      | ^~~~~~~~~~~~~~
      | %define parse.error verbose
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_parser.y: warning: fix-its can be applied.  Rerun with option '--update'. [-Wother]
[ 50%] [FLEX][VqmLexer] Building scanner with flex 2.6.4
[ 50%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_common.c.o
[ 50%] Building CXX object abc/CMakeFiles/libabc.dir/src/sat/glucose/Options.cpp.o
[ 50%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_dll.cpp.o
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_dll.cpp: In function ‘int vqm_get_error_message(char*, int)’:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/libs/libvqm/vqm_dll.cpp:120:23: warning: passing argument 1 to ‘restrict’-qualified parameter aliases with argument 2 [-Wrestrict]
  120 |                 strcpy(message_buffer, message_buffer);
      |                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 50%] Building CXX object libs/libvqm/CMakeFiles/libvqm.dir/vqm_lexer.gen.c.o

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/vqm2blif/src/base/hard_block_recog.cpp:1738:31: warning: unknown escape sequence: '\]'
 1738 |     std::regex port_is_a_bus ("(.*)[[]([0-9]*)\]~(?:.*)");
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
[ 85%] Building CXX object utils/vqm2blif/CMakeFiles/libvqm2blif.dir/src/base/hash.cpp.o
[ 85%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/base/check_netlist.cpp.o
[ 85%] Building CXX object utils/vqm2blif/CMakeFiles/libvqm2blif.dir/src/base/lut_recog.cpp.o
[ 85%] Building CXX object utils/vqm2blif/CMakeFiles/libvqm2blif.dir/src/base/lut_stats.cpp.o
[ 85%] Building CXX object utils/vqm2blif/CMakeFiles/libvqm2blif.dir/src/base/preprocess.cpp.o
[ 85%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/base/clock_modeling.cpp.o
[ 85%] Building CXX object utils/vqm2blif/CMakeFiles/libvqm2blif.dir/src/base/vqm2blif_util.cpp.o
[ 85%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/base/clustered_netlist.cpp.o
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/vqm2blif/src/base/vqm2blif_util.cpp: In function ‘void verify_hard_block_type_name(std::string)’:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/vqm2blif/src/base/vqm2blif_util.cpp:63:47: warning: unknown escape sequence: '\$'
   63 |     std::regex verilog_VHDL_naming_rules_one ("^[a-zA-Z_][a-zA-Z_\$0-9]*[a-zA-Z_\$0-9]$");
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/vqm2blif/src/base/vqm2blif_util.cpp:63:47: warning: unknown escape sequence: '\$'

#### Possible Solution
Fix the warnings.

#### Your Environment
Occurred in a CI run on April 24, 2023:

https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4790484294/jobs/8519669398?pr=2293

AlexandreSinger, 2024-04-04:
@kimiatkh As with issue #2403 , what is the progress on this issue?

kimiatkh, 2024-04-17:
@AlexandreSinger My apologies for the delayed response. I haven't had the chance to work on this yet. But I'll do my best to address this by next week. I'll keep you posted on any updates regarding the issue.

AlexandreSinger, 2024-04-17:
@kimiatkh Not a problem at all! There is no rush on this, I have just been eating away at these warnings over the last couple of weeks. I have just raised an PR on the regex warnings (PR #2537 ). The strcpy one is the only one left I think. I have been talking with Soheil; we think that this one may actually be a bug where the wrong value was being copied. Whenever you get the chance to look into it, or if you are low on time, do let me know!

AlexandreSinger, 2024-05-16:
I believe all of these warnings are now gone after Navid's most PR. PR #2549 

Shall we close this issue @vaughnbetz

AlexandreSinger, 2025-05-22:
This issue has been resolved. Closing.