verilog-to-routing/vtr-verilog-to-routing #2295: Odin II and blifexplorer compiler warnings (should fix)

vaughnbetz, 2023-04-25:
#### Expected Behaviour
We should have no compiler warnings.

#### Current Behaviour
Odin II and blifexplorer and have compiler warnings.

Some are:  

 /usr/include/c++/11/type_traits:2960:14: warning: noexcept-expression evaluates to ‘false’ because of a call to ‘resolve_binary_operation(ast_node_t*)::<lambda(long int)>’ [-Wnoexcept]
 2960 |       return noexcept(std::declval<_Fn>()(std::declval<_Args>()...));
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/odin_ii/src/ast/ast_loop_unroll.cpp:323:16: note: but ‘resolve_binary_operation(ast_node_t*)::<lambda(long int)>’ does not throw; perhaps it should be declared ‘noexcept’
  323 |         return [=](long value) {
      |                ^

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp:275:37: warning: ‘QMatrix QGraphicsView::matrix() const’ is deprecated: Use transform() [-Wdeprecated-declarations]
  275 |     QMatrix oldMatrix = view->matrix();
      |                         ~~~~~~~~~~~~^~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QtWidgets:135,
                 from /home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp:24:
/usr/include/x86_64-linux-gnu/qt5/QtWidgets/qgraphicsview.h:169:48: note: declared here
  169 |     QT_DEPRECATED_X("Use transform()") QMatrix matrix() const;
      |                                                ^~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp:276:22: warning: ‘void QGraphicsView::resetMatrix()’ is deprecated: Use resetTransform() [-Wdeprecated-declarations]
  276 |     view->resetMatrix();
      |     ~~~~~~~~~~~~~~~~~^~
In file included from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QtWidgets:135,
                 from /home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp:24:
/usr/include/x86_64-linux-gnu/qt5/QtWidgets/qgraphicsview.h:171:50: note: declared here
  171 |     QT_DEPRECATED_X("Use resetTransform()") void resetMatrix();
      |                                                  ^~~~~~~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp: In member function ‘void MainWindow::openFileWithOdin()’:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp:928:37: warning: ‘constexpr QFlags<T>::QFlags(QFlags<T>::Zero) [with Enum = QMessageBox::StandardButton; QFlags<T>::Zero = int QFlags<QMessageBox::StandardButton>::Private::*]’ is deprecated: Use default constructor instead [-Wdeprecated-declarations]
  928 |                            , 0, this);
      |                                     ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:1299,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QtCore:4,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QtWidgetsDepends:3,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QtWidgets:3,
                 from /home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/mainwindow.cpp:24:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qflags.h:123:80: note: declared here
  123 |     QT_DEPRECATED_X("Use default constructor instead") Q_DECL_CONSTEXPR inline QFlags(Zero) noexcept : i(0) {}
      |                                                                                ^~~~~~

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/blifexplorer/src/container.cpp:943:29: warning: ‘int qrand()’ is deprecated: use QRandomGenerator instead [-Wdeprecated-declarations]
  943 |              int act = qrand() % 255;

#### Possible Solution
Get rid of the deprecated functions etc.

#### Your Environment
April 24, 2023 CI run:  https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4790484294/jobs/8519669398?pr=2293

alirezazd, 2023-05-05:
@navidjafarof Please take a look at this,  should be easy to fix.

vaughnbetz, 2025-05-26:
I believe this was fixed and should be closed. @AlexandreSinger : if we still have warnings to get rid of please let me know.

closed 2025-05-26 by vaughnbetz