verilog-to-routing/vtr-verilog-to-routing #2296: Fix Fasm warnings

vaughnbetz, 2023-04-25:
#### Expected Behaviour
We should have no compiler warnings

#### Current Behaviour
Fasm has a few warnings

/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp:15:28: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
   15 |         CHECK(table.size() == (1 << num_inputs));
      |               ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp: In function ‘void {anonymous}::CATCH2_INTERNAL_TEST_2()’:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp:30:28: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
   30 |         CHECK(table.size() == (1 << num_inputs));
      |               ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp: In function ‘void {anonymous}::CATCH2_INTERNAL_TEST_4()’:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp:45:32: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
   45 |             CHECK(table.size() == (1 << num_inputs));
      |                   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp: In function ‘void {anonymous}::CATCH2_INTERNAL_TEST_6()’:
/home/runner/work/vtr-verilog-to-routing/vtr-verilog-to-routing/utils/fasm/test/test_lut.cpp:61:32: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
   61 |             CHECK(table.size() == (1 << num_inputs));
      |                   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~

Also have a QoR failure: if this recurs, please update the golden results.

[Fail]
k4_n4_v7_bidir.xml/styr.blif/common critical_path_delay relative value 1.534219922694261 outside of range [0.5,1.4] and not equal to golden value: 9.23088
[Fail]
k4_n4_v7_bidir.xml/styr.blif/common geomean_nonvirtual_intradomain_critical_path_delay relative value 1.534219922694261 outside of range [0.5,1.4] and not equal to golden value: 9.23088
[Fail]
k4_n4_v7_bidir.xml/styr.blif/common setup_WNS relative value 1.534219922694261 outside of range [0.5,1.4] and not equal to golden value: -9.23088

#### Possible Solution
Fix them.

#### Context

#### Your Environment
April 24 CI build: https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/4790484294/jobs/8519669398?pr=2293

vaughnbetz, 2024-03-27:
I'm not sure ... I'll take a look when I get a chance.

AlexandreSinger, 2025-05-23:
Closing this issue before the stale issue tracker gets to it. This has been resolved as far as I can tell from the CI.

closed 2025-05-23 by AlexandreSinger