verilog-to-routing/vtr-verilog-to-routing #2302: Using Parmys For Xilinx-Like Architectures

amin1377, 2023-05-04:
I tried to implement `stereovision0.v` on `simple-7series.xml`. I ran this command: `run_vtr_flow.py stereovision0.v simple-7series.xml` and got this error message:
`simple-7series/stereovision0		Error: Executable yosys failed full command:  /usr/bin/env time -v /home/mohagh18/vtr-verilog-to-routing/build/bin/yosys -c synthesis.tcl returncode  :  1 log file    :  temp/parmys.out failed: Executable yosys failed (took 0.27 seconds, overall memory peak 11.65 MiB consumed by parmys run)`

[Here](https://utoronto-my.sharepoint.com/:u:/g/personal/amin_mohaghegh_mail_utoronto_ca/EVCYULZBos9KrZRaubFQob4BnaAz5DP78VRQon3lNrTPSQ?e=XlHIYm) is the `parmys.out` file.

WhiteNinjaZ, 2023-05-05:
It looks like this is a synthesis problem with the Parmys[Yosys] front end not an arch issue. I am getting the same problem on multiple old example architectures including the example arch from the documentation (i.e. `$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py $VTR_ROOT/doc/src/quickstart/blink.v $VTR_ROOT/vtr_flow/arch/timing/EArch.xml -temp_dir . --route_chan_width 100`). Odin works fine, but when trying to use yosys front end I am getting the same issue as amin.

vaughnbetz, 2023-05-05:
Thanks Joshua. @alirezazd any ideas?

poname, 2023-05-05:
I just tested `stereovision0.v` on `simple-7series.xml` and it was OK.
The command history:
```
  504  git clone https://github.com/verilog-to-routing/vtr-verilog-to-routing.git vtr
  505  cd vtr
  506  make -j10
  507  cd vtr_flow/scripts/
  508  ./run_vtr_flow.py ../benchmarks/verilog/stereovision0.v ../arch/xilinx/simple-7series.xml
```

and the output:
```
simple-7series/stereovision0            OK (took 1250.60 seconds, overall memory peak 597.01 MiB consumed by vpr run)

```

`Yosys` not getting along with `temp_dir` was reported previously in #2152.

poname, 2023-05-05:
@alirezazd we can have one of the students (Navid or Ritwik) working on the `temp_dir` issue.

alirezazd, 2023-05-05:
@amin1377 @WhiteNinjaZ  Please check if the workaround in the mentioned issue works. You should only add temp dir path to your command : ./run_vtr_flow PATH_TO_VERILOG_FILE.v PATH_TO_ARCH_FILE.xml -start yosys -temp_dir /full/path/to/temp_dir

@poname Thank you for mentioning this.

WhiteNinjaZ, 2023-05-05:
@alirezazd can confirm that adding full path does fix the issue (`$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py     $VTR_ROOT/doc/src/quickstart/blink.v     $VTR_ROOT/vtr_flow/arch/timing/EArch.xml     -temp_dir /full/path     --route_chan_width 100`). @amin1377 I am getting failures in vtr when trying to run tests on simple-7series_diagonal.xml but that actually could be an arch issue (ill look into that further). Simple-7series and simple-7series_correctedSB appear to be working though at least on the smaller designs without multipliers.

alirezazd, 2023-05-05:
@WhiteNinjaZ @amin1377 Happy that its resolved. Its a simple workaround that should be sufficient enough for now. I will add this issue to our list to track and resolve it in the future since we are currently working on two critical issues.

farukyld, 2024-03-26:
I get similar error when running this command:
```bash
$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py \
    $VTR_ROOT/doc/src/quickstart/blink.v \
    $VTR_ROOT/vtr_flow/arch/timing/EArch.xml \
    -temp_dir . \
    --route_chan_width 100
```
inside another directory as instructed in the [documentation](https://docs.verilogtorouting.org/en/latest/quickstart/#:~:text=Next%2C%20make%20a%20new%20directory%20to%20work%20in%20named).

the error in the console:
```bash
EArch/blink             Error: Executable yosys failed
        full command:  /usr/bin/env time -v /home/usr1/vtr-verilog-to-routing/build/bin/yosys -c synthesis.tcl
        returncode  :  1
        log file    :  parmys.out
failed: Executable yosys failed (took 0.02 seconds, overall memory peak 65.64 MiB consumed by vpr run)
```

This was the related lines (I guess) in the parmys.out file:
```
ERROR: Parmys Failed to load architecture file: Failed to open file with exit code ERROR_PARSE_ARCH at line: -1
Command exited with non-zero status 1
	Command being timed: "/home/usr1/vtr-verilog-to-routing/build/bin/yosys -c synthesis.tcl"
	User time (seconds): 0.00
	System time (seconds): 0.00
	Percent of CPU this job got: 100%
	Elapsed (wall clock) time (h:mm:ss or m:ss): 0:00.00
	Average shared text size (kbytes): 0
	Average unshared data size (kbytes): 0
	Average stack size (kbytes): 0
	Average total size (kbytes): 0
	Maximum resident set size (kbytes): 14348
	Average resident set size (kbytes): 0
	Major (requiring I/O) page faults: 0
	Minor (reclaiming a frame) page faults: 768
	Voluntary context switches: 1
	Involuntary context switches: 0
	Swaps: 0
	File system inputs: 0
	File system outputs: 0
	Socket messages sent: 0
	Socket messages received: 0
	Signals delivered: 0
	Page size (bytes): 4096
	Exit status: 1
```

I was able to resolve this by 
```python
    architecture_file_path = str(vtr.paths.scripts_path / architecture_file)
```
with this:
```python
architecture_file_path = str(architecture_file)
```
in this file: `vtr-verilog-to-routing/vtr_flow/scripts/python_libs/vtr/parmys/parmys.py` line `212` at this commit:

#23535b02bb3afde26daf9d3887606c9e21192311

vaughnbetz, 2024-03-26:
Thanks for this report. @alirezazd can you update this or assign to someone else on the UNB team to do so?  We definitely want the quick start to work with no issues.

alirezazd, 2024-03-26:
@vaughnbetz I am assigning this to our new PhD student, Amir. Feel free to ask him for updates.

vaughnbetz, 2025-05-26:
@WhiteNinjaZ : is this resolved now?  If it is please close it, and if not, what do we need to do?

closed 2025-05-27 by WhiteNinjaZ