verilog-to-routing/vtr-verilog-to-routing #2318: About the suspected anomaly in power estimation

jdzhu19, 2023-05-21:
#### Expected Behaviour
softmax.v should be a relatively DSP-heavy design, so when the architecture is `k6_ N10_ I40_ Fi6_ L4_ frac0_ ff1_ 22nm.xml`, the benchmark is `softmax.v`, the power of mult_ 36 should not be so low.

#### Current Behaviour
```
When the architecture is `k6_ N10_ I40_ Fi6_ L4_ frac0_ ff1_ 22nm.xml`, the benchmark is `softmax.v`, the power report file displays as follows: 
mult_36                         3.535e-06   1.656e-05   1           Sum of Children
 Mode:two_divisible_mult_18x18  3.535e-06   1.656e-05   1           
  divisible_mult_18x18          3.535e-06   1.656e-05   1           Sum of Children
   Mode:two_mult_9x9            0           0           -nan        
    mult_9x9_slice              0           0           -nan        Pin-Toggle  
   Mode:mult_18x18              3.535e-06   1.656e-05   1           
    mult_18x18_slice            3.535e-06   1.656e-05   1           Pin-Toggle  
 Mode:mult_36x36                0           0           -nan        
  mult_36x36_slice              0           0           -nan        Pin-Toggle  
```

#### Steps to Reproduce
Open the power estimation option and run the vtr process with `k6_ N10_ I40_ Fi6_ L4_ frac0_ ff1_ 22nm.xml` and `softmax.v`.

vaughnbetz, 2023-05-23:
Jeff could you or someone on your team take a look?

jdzhu19, 2023-05-30:
Hi! I just wanted to check if this question has been addressed or if there is already an answer.

jdzhu19, 2023-06-13:
> Jeff could you or someone on your team take a look?

Hello! It's been two weeks and this question is still bothering me and I wanted to make sure you have been paying attention to it. Thank you very much for your time!

closed 2025-06-08 by github-actions[bot]