verilog-to-routing/vtr-verilog-to-routing #2322: Allow pin offsets (x, y, layer) in connection block specifications to build the rr-graph

vaughnbetz, 2023-05-24:
Currently vpr allows the pins connecting blocks -> routing channel segments to be on various sides and locations within or on the perimeter of a block. Xilinx 7-series (and perhaps Achronix devices) however allow a block to connect to a different set of routing wire segments: those to the left and above a block, but also a horizontal channel at x-1 and a vertical channel at y+1.  

#### Proposed Behaviour
1. We could capture this behaviour by allowing the side + pin-offset of blocks to be outside a block (e.g. x-1, y+1). When applied to a layer, this offset notation would also enable us to connect between layers.
2. I believe Xilinx drives left-going (decreasing) wires at the x-1 and y locations, and increasing (right-going) wires at x and y+1. Ideally we'd have an option to force this behaviour too as it is logical.  This could be considered a separate enhancement though, where we can specify a type of wire (inc/dec) along with the Fc value and pin location for a pin.

#### Current Behaviour
1. I believe we can't specify a pin that is outside the area of a block.
2. We can't specify connect to only increasing or decreasing wires.

#### Context
We can bring in a custom rr-graph, but this limitation makes it harder to explore xilinx-like architectures.

WhiteNinjaZ, 2024-06-18:
@vaughnbetz sorry for the late reply, I was going to talk about this in the VTR meeting but forgot too. Yes this would be very useful to us and give a much cleaner arch.

amin1377, 2025-06-18:
@vaughnbetz: Regarding this issue, I think once OpenFPGA is merged, the VIB approach ([Link](https://ieeexplore.ieee.org/document/10416125) to the paper) might help. Based on what they describe, it integrates all tile interconnects (both CB and SB) into a single Versatile Interconnect Block (VIB). In theory, this should address the problem, although I haven’t tested it yet.

vaughnbetz, 2025-06-18:
Thanks. It would be great if you could test this out.