verilog-to-routing/vtr-verilog-to-routing #2347: Cannot Run "run_vtr_flow" With Parmys

amin1377, 2023-07-07:
I am attempting to utilize run_vtr_flow to execute the complete VTR flow, which includes synthesis and VPR stages. When I specify odin as the synthesizer (with the following parameter: -start odin), I encounter no issues. However, when I do not specify any synthesizer (which defaults to parmys), an error occurs indicating the inability to locate the architecture file. This is the error:

`1. Executing Verilog-2005 frontend: /home/mohagh18/Desktop/vtr-verilog-to-routing/build/bin/../share/yosys/parmys/vtr_primitives.v
Using parmys as partial mapper
ERROR: Odin Failed to load architecture file: Failed to open file with exit code ERROR_PARSE_ARCH at line: -1
`

@vaughnbetz

vaughnbetz, 2023-07-08:
@alirezazd can you take a look or reassign please?

alirezazd, 2023-07-10:
@amin1377 Have you tired building the flow with Odin enabled? You should build the flow with the respective flag:

make CMAKE_PARAMS="-DWITH_ODIN=ON"

https://docs.verilogtorouting.org/en/latest/odin/

amin1377, 2023-07-10:
@alirezazd: I don't encounter any issues when using ODIN as the synthesizer. However, I face a problem when attempting to use parmys as the synthesizer. In that case, run_vtr_flow encounters a problem with the architecture file.

alirezazd, 2023-07-11:
@amin1377 Probably your architecture file contains blocks that are not supported by Odin. @poname Any ideas?

amin1377, 2023-07-11:
@alirezazd: But when I use run_vtr_task to run the whole task, I don't face any problem.

poname, 2023-07-11:
I guess this may be related to #2152 and https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2232#issuecomment-1408966619.
If you are using `-temp_dir` option, it requires a full path.

Otherwise, please provide more information on the input files and the specific command.

amin1377, 2023-07-13:
@poname I am not using `-temp_dir`. I faced the error that I mentioned whenever I am running `run_vtr_flow`. 
For instance, I first copied `vtr_flow/benchmarks/verilog/stereovision3.v` and `vtr_flow/arch/timing/k6_frac_N10_frac_chain_mem32K_40nm.xml` to a directory, then ran the following command: `$VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py stereovision3.v k6_frac_N10_frac_chain_mem32K_40nm.xml --route_chan_width 100` and got the following error:
```
k6_frac_N10_frac_chain_mem32K_40nm/stereovision3		Error: Executable yosys failed
	full command:  /usr/bin/env time -v /home/mohagh18/vtr-verilog-to-routing/build/bin/yosys -c synthesis.tcl
	returncode  :  1
	log file    :  temp/parmys.out
failed: Executable yosys failed (took 0.10 seconds, overall memory peak 13.33 MiB consumed by parmys run)
```
This is the message in `parmys.out` file:
```
1. Executing Verilog-2005 frontend: /home/mohagh18/vtr-verilog-to-routing/build/bin/../share/yosys/parmys/vtr_primitives.v
Using parmys as partial mapper
ERROR: Odin Failed to load architecture file: Failed to open file with exit code ERROR_PARSE_ARCH at line: -1
Command exited with non-zero status 1
```

poname, 2023-07-13:
@amin1377 Do you see these two folders in `$VTR_ROOT/build/share/yosys/`, 
and their contents should match the picture:

[image]

amin1377, 2023-07-13:
@poname Yes, I can see all the files. I have also attached the content of the directories that you mentioned.

[image]

poname, 2023-07-14:
Somehow unexpected. Did the same and got no errors!
```
[dev@dev baag]$ /home/dev/vtr-verilog-to-routing/vtr_flow/scripts/run_vtr_flow.py stereovision3.v k6_frac_N10_frac_chain_mem32K_40nm.xml --route_chan_width 100
k6_frac_N10_frac_chain_mem32K_40nm/stereovision3                OK (took 2.16 seconds, overall memory peak 68.78 MiB consumed by vpr run)
```
What is the CMake version on your machine? 
Are all dependencies installed according to the [install_apt_packages.sh](https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/81a4a8e7f2bc963927eb7677f7b146f4530bdf1a/install_apt_packages.sh) script?
And I suggest using the full path instead of `$VTR_ROOT` to see if it helps.
@amin1377

amin1377, 2023-07-17:
My CMack version is 3.26.4. I've already run the script you mentioned. I attempted using the full path, but unfortunately, it didn't resolve the issue.

Additionally, I tested the script on another system with cmake version 3.26.0, and encountered the same error.

amin1377, 2023-07-17:
@saaramahmoudi @MohamedElgammal @duck2 Have any of you encountered the same issue when running run_vtr_flow, or is it specific to my case only?

saaramahmoudi, 2023-07-18:
@amin1377 Yes, I have the same issue when running run_vtr_flow.

alirezazd, 2023-07-18:
@poname I will run it and report back here.

alirezazd, 2023-07-18:
@poname @amin1377 @saaramahmoudi 

I just ran the flow without any problems:
[image]

I'm pretty sure that your issue rises from a missing dependecy, problematic build config or other common problems that can lead to this and are solely related to your local config.

alirezazd, 2023-07-18:
I have extracted my local config with a script. Both the script and the extracted build info are attached.
you can try running the script and get a diff of your config against mine to further check your issue.
[build_info.zip](https://github.com/verilog-to-routing/vtr-verilog-to-routing/files/12083790/build_info.zip)

huoshenlaile, 2023-07-20:
Hi, I have the exact same issue.
TLDR: I solve this by explicitly specifying the full path of `temp_dir` like this:
```
python3 $VTR_ROOT/vtr_flow/scripts/run_vtr_flow.py blink.sv EArch.xml -temp_dir /home/user/vtr_work/quickstart/blink_sv_flow/temp -parser system-verilog 
```
and then it worked.

More details:
I confirmed that I did not compile the odin, and I manually checked that odin executable is not installed. But the error message still shows, it is misleading as it did not actually use odin. There is some small issue with the temp_dir for parmys, if you dont want to manually type the whole path of temp_dir every time, you can modify the `run_vtr_flow.py` by
1. first add `import os`
2. change parse `-temp_dir` around line 187 to this:
``` python
    house_keeping.add_argument(
        "-temp_dir",
        default=os.getcwd() + "/temp",
        help="Directory to run the flow in (will be created if non-existant).",
    )
```

amin1377, 2023-07-24:
@huoshenlaile Thanks! The solution you mentioned (using absolute dir) solved my problem.

amin1377, 2023-07-24:
@vaughnbetz: I will make a PR first to fix the issue as @huoshenlaile suggested and then will close this issue.

vaughnbetz, 2023-07-24:
Sounds good; thanks all.

closed 2023-09-12 by amin1377