verilog-to-routing/vtr-verilog-to-routing #2352: Can't write out large rr_graphs in capnp format

duck2, 2023-07-23:
`direct_rf` from `titan_quick_qor` with default settings creates a rr_graph with ~236M edges. Trying to write it out will result in a crash with the message `failed: total size of struct list is larger than max segment size`.

#### Expected Behaviour
If the rr_graph fits in memory, we should be able to write it out to the disk.

#### Current Behaviour
We can't

#### Possible Solution
Apparently Cap'n Proto has a hard limit [[1]](https://groups.google.com/g/capnproto/c/H4fao91RFH8) [[2]](https://capnproto.org/encoding.html) for words in a single _segment_. This also applies to lists of structs. From [2]:

> D (29 bits) = Size of the list:
>   when C <> 7: Number of elements in the list.
>    when C = 7: Number of words in the list, not counting the tag word
>    (see below).
> [...]
> A struct list must always be written using C = 7.

In my understanding, this means the size of a list of structs is limited to 8 bytes * 2^29 = 2 GiB. I think the largest rr_graph I've seen was around 4 GB so it checks out.

Any ideas about how to go about this? I can't think of a way to solve it without manually tweaking the schema, which somewhat ruins the purpose of having a auto-generated one.

#### Steps to Reproduce
1. Run `titan_quick_qor`
2. Go to run directory for `directrf_stratixiv_arch_timing.blif`
3. Get run command from `vpr.out`, append `--write_rr_graph rr_graph.bin` and execute it

duck2, 2023-07-24:
@amin1377 @kimiatkh Any clue about this?

amin1377, 2023-07-24:
> @amin1377 @kimiatkh Any clue about this?
I have never used capnp format to serialize the RR graph. So far, I have only used XML to serialize it, and it has been faster than building the graph from scratch (at least for my test cases which usually involve big circuits)