verilog-to-routing/vtr-verilog-to-routing #2361: Add some overview to the RRGraphView doxygen documentation

vaughnbetz, 2023-08-03:
#### Proposed Behaviour
We should give a high level overview of the RRGraphView data structure / API in our doxygen documentation. 
#### Current Behaviour
Currently we dive right into methods.

#### Possible Solution
Explain the RRGraph describes the programmable routing fabric of the FPGA (all wires and switches). RRGraphView provides a read-only interface, which tools like the router, graphics, and statistics need.
Should also explain that the fundamental data structures is a graph (nodes and out-going edges) with extra information like the location within the chip and electrical parameters added to help make algorithms more efficient, draw the chip, and estimate delays.

vaughnbetz, 2024-06-13:
Re assigning to @tangxifan . This might be a duplicate though!

vaughnbetz, 2024-06-13:
Is a duplicate of the remaining work in https://github.com/verilog-to-routing/vtr-verilog-to-routing/issues/1834 . We can close both together.

tangxifan, 2024-06-13:
@vaughnbetz Indeed. We may close this one and keep #1834 open. My team will follow up. A PR will happen this summer.

vaughnbetz, 2025-06-14:
This has been done; closing.

closed 2025-06-14 by vaughnbetz