verilog-to-routing/vtr-verilog-to-routing #2368: VTR NoC Documentation Clean-up

Srivat97, 2023-08-14:
This issue will be a place holder to report problems with the newly added NoC related documentation. Please comment in this issue any found mistakes or changes you feel are needed. Once I fix a mistake or make a change I will mark comment as **"fixed"**. You can go ahead fix them as well, just make sure to update the comment.

Below are pointers to the NoC documentation:
NoC Code Documentation: https://docs.verilogtorouting.org/en/latest/api/vprinternals/vpr_noc/
NoC Architecture Description: https://docs.verilogtorouting.org/en/latest/arch/reference/#noc-description
NoC Command line Parameters: https://docs.verilogtorouting.org/en/latest/vpr/command_line_usage/#noc-options
NoC Traffic Flow file format: https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#network-on-chip-noc-traffic-flows-format-flows

Listing initial documentation reviewers here (feel free to add more reviewers):
@saaramahmoudi 
@vaughnbetz 
@Srivat97

vaughnbetz, 2023-08-18:
This all looks very good to me. I quickly looked at each tab and they look fine.

closed 2025-06-10 by github-actions[bot]

vaughnbetz, 2025-06-10:
This was completed.