verilog-to-routing/vtr-verilog-to-routing #2395: CI Test Failure

amin1377, 2023-09-14:
The CI tests are currently failing. The error it displays is "Failed to resolve the action download info. Error: Temporary Redirect." One thing I suspect that may be causing this issue is the changes made in this [PR](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2390), where I changed the direction of the directory created by "run_vtr_flow.py" from a relative path to an absolute path. I attempted to address this by reverting those changes in this [PR](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2394), but that did not resolve the issue.

Any insights into the cause of these failures?
@duck2 
@mithro 
@hzeller 
@acomodi 
@vaughnbetz

vaughnbetz, 2023-09-14:
It looks like the github runners are working, but the google cloud is refusing to download things.

Google cloud machine: (set up job part):
17:07:05 | Current runner version: '[2](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801141100?pr=2394#step:2:2).309.0'
17:07:05 | Current coordinator sha: '4dd6a0c82bffd25[3](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801141100?pr=2394#step:2:3)a64a158f597e6e88e2c81ed9'
17:07:05 | Runner name: 'gh-actions-runner-vtr_12'
17:07:05 | Runner group name: 'Default'
17:07:05 | Machine name: 'gh-actions-runner-vtr'
17:07:05 | Prepare workflow directory
17:07:05 | Prepare all required actions
17:07:05 | Getting action download info
17:07:06 | Failed to resolve action download info. Error: Temporary Redirect
17:07:06 | Retrying in 28.011 seconds
17:07:3[4](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801141100?pr=2394#step:2:4) | **Failed to resolve action download info. Error: Temporary Redirect**
17:07:34 | Retrying in 18.[5](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801141100?pr=2394#step:2:5)1[9](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801141100?pr=2394#step:2:9) seconds

Github runner: (set up job part):
Current runner version: '[2](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801132684?pr=2394#step:1:2).309.0'
Operating System
Runner Image
Runner Image Provisioner
GITHUB_TOKEN Permissions
Secret source: Actions
Prepare workflow directory
Prepare all required actions
**Getting action download info
Download action repository 'actions/setup-python@v4' (SHA:[6](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801132684?pr=2394#step:1:7)1a6322f88396a6271a6ee3565807d608ecaddd1)
Download action repository 'actions/checkout@v3' (SHA:f43a0e5ff2bd294095638e18286ca9a3d1956744)
Download action repository 'hendrikmuhs/ccache-action@v1.2' (SHA:ca3acd2731eef[11](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801132684?pr=2394#step:1:13)f1572ccb126356c2f9[29](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801132684?pr=2394#step:1:33)8d35e)
Download action repository 'actions/upload-artifact@v3' (SHA:a8a3f3ad[30](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801132684?pr=2394#step:1:34)e3422c9c7b888a15615d19a852ae[32](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6188658305/job/16801132684?pr=2394#step:1:36))**
Complete job name: R: Basic

@hermanschmit : any ideas on what might cause the google cloud runners to fail, or any pointers toward someone who might know?  @duck2: you tend to be good at debugging CI -- any ideas?

amin1377, 2023-09-14:
This is the message shown for the CI tests that passed before:

14:13:27 | Current runner version: '2.309.0'
14:13:[2](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6160670868/job/16718173853#step:2:2)7 | Current coordinator sha: '4dd6a0c82bffd253a64a158f597e6e88e2c81ed9'
14:13:27 | Runner name: 'gh-actions-runner-vtr_16'
14:1[3](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6160670868/job/16718173853#step:2:3):27 | Runner group name: 'Default'
14:13:27 | Machine name: 'gh-actions-runner-vtr'
1[4](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6160670868/job/16718173853#step:2:4):13:27 | Prepare workflow directory
14:13:27 | Prepare all required actions
14:13:27 | Getting action download info
14:13:27 | Download action repository 'actions/checkout@v3'
14:13:2[8](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6160670868/job/16718173853#step:2:8) | Download action repository 'actions/upload-artifact@v3'

Based on the [GitHub status](https://www.githubstatus.com/) and _checkout_ action [issue](https://github.com/actions/checkout/issues) pages, it seems this is a Google server issue.

@kgugala : and ideas?

vaughnbetz, 2023-09-14:
@kimiatkh @duck2 : another thing to check: any recent change to where we download titan or TBB from?  Checking if a change to the server for one of those could be causing the failure (seems less likely than a google cloud / server permission issue though given the message).

kimiatkh, 2023-09-14:
@vaughnbetz I have updated the URL address for the Titan archive, but my changes are still on the branch. I don't believe there have been any changes related to that on the master branch.

vaughnbetz, 2023-09-15:
Thanks Kimia.  That eliminates another possibility.

amin1377, 2023-09-15:
@vaughnbetz It seems the problem is solved ([link](https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6199750583/job/16832974393)). It is now passing the "Set up job" step.

vaughnbetz, 2023-09-15:
Yaay!

amin1377, 2023-09-15:
@vaughnbetz if it's alright with you, I would like to keep this issue open for a few more days, just in case. If you think it is unnecessary, I guess you should be able to close the issue or let me know to close it.

vaughnbetz, 2023-09-15:
Sure, sounds good.

closed 2023-09-24 by amin1377