verilog-to-routing/vtr-verilog-to-routing #2402: Warnings and failure in vtr_reg_nightly_test2_odin

vaughnbetz, 2023-09-28:
We have some compilation warnings and a failure in the CI test run at https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6333593038/job/17201935175

#### Expected Behaviour
We should be warning clean and the test should pass.

#### Current Behaviour
Warnings that should be fixed:

02:04:36 | [ 81%] Building CXX object utils/vqm2blif/CMakeFiles/libvqm2blif.dir/src/base/hard_block_recog.cpp.o
02:04:36 | /root/vtr-verilog-to-routing/vtr-verilog-to-routing/odin_ii/src/verilog/verilog_bison.y: warning: 2 shift/reduce conflicts [-Wconflicts-sr]
02:04:36 | /root/vtr-verilog-to-routing/vtr-verilog-to-routing/odin_ii/src/verilog/verilog_bison.y: note: rerun with option '-Wcounterexamples' to generate conflict counterexamples

The test then runs and has 0 failures, but the overall regtest is still flagged as a failure in CI. The only thing that seems to go wrong is that a git submodule under parmys doesn't seem to download:

During build:
02:06:24 | [ 95%] Building CXX object vpr/CMakeFiles/libvpr.dir/src/util/vpr_utils.cpp.o
02:06:26 | fatal: not a git repository: './/.git'
02:06:27 | [ 95%] Built target yosys

During post-run actions:

07:54:25 | [command]/usr/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :"
07:54:26 | fatal: No url found for submodule path 'parmys/third_party/googletest' in .gitmodules
Warning: The process '/usr/bin/git' failed with exit code 128

alirezazd, 2023-10-04:
@poname please take a look at this

poname, 2023-10-05:
@vaughnbetz 
Regarding the `git submodule` issue, I think this is some flaky behaviour related to the infrastructure and perhaps networking issues, since everything is green in the scheduled task at https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/6413159440.

However, I have created a PR to disable Parmys in Odin-specific tests like `vtr_reg_nightly_test2_odin` to reduce the flakiness probability.

I'll let you know once it's ready to be merged.

poname, 2023-10-05:
The warnings are related to `shift/reduce conflicts` within the `verilog_bison.y` file and have to be inspected separately.
@alirezazd

AlexandreSinger, 2024-04-04:
@alirezazd I am collecting all of the warnings in VTR (see issue #2518 ). Has there been any progress on resolving these issues?

alirezazd, 2024-04-05:
@AlexandreSinger Currently, I'm away and have a full plate with other tasks, making it difficult for me to address this issue promptly. For this and future updates, could you please reach out to @amirarjmand93? They will be able to assist you further and provide the necessary support.
@amirarjmand93 please keep us informed with your progress.

Thank you for your understanding.

alirezazd, 2024-04-05:
@AlexandreSinger Just a quick reminder, Odin's parser is not being maintained anymore since we are using Yosys as the main parser for the frontend. According to @poname the problem roots back to `verilog_bison.y` so it is clearly a parser issue. I would suggest ignoring this warning for now until we finalize deprecating  Odin.

vaughnbetz, 2024-04-06:
One way to make this warning go away without changing the grammar is to use the %expect n statement to say we expect some conflicts like this.  The conflicts themselves aren't dangerous; the grammar is ambiguous but bison interprets it in a specified way which clearly is working. So I think just marking it as an expected conflict with this syntax would be a good fix.

I think we should deprecate odinII slowly; it should be kept around through VTR 9.0 so anyone depending on it has time to adapt. So I think making this warning go away (using this low effort method) would be good. See the link below for details.
https://www.gnu.org/software/bison/manual/html_node/Shift_002fReduce.html

amirarjmand93, 2024-04-17:
@AlexandreSinger I am trying to reach out to the warning but I am dealing with a problem when I run the test. 

this is my path :

1- i did make Odin-ii by _make CMAKE_PARAMS="-DWITH_ODIN=ON"_ and all dependencies were installed. 
2- i used _<scripts_path>/run_vtr_task.py -l <tasks_path>/regression_tests/vtr_reg_nightly_test_2_odin/task_list.txt_
3- have got error -> _Error: Failed to resolve VTR source file carpat_stratixiv_arch_timing.blif_
but some of the tests(except  vtr_reg_nightly_test_2_odin) in the _regression_tests_ folder passed well.

can you help me reproduce the path to reaching the warnings you got?

AlexandreSinger, 2024-04-17:
@amirarjmand93 

The warnings we are seeing can be found in the CI here (this is from the most recent CI build of master): https://github.com/verilog-to-routing/vtr-verilog-to-routing/actions/runs/8714477843/job/23904790647#step:6:1738

It occurs at build time, not at run time, so you do not need to run the nightly tests to replicate the warning.

make params that cause this issue can be found at the start of the test:
[image]

So, I think to replicate this issue, you would just need to perform a clean build and build with the same CMAKE_PARAMS:
```
make clean
make CMAKE_PARAMS="-DVTR_ASSERT_LEVEL=3 -DVTR_ENABLE_SANITIZE=on -DVTR_IPO_BUILD=off -DWITH_BLIFEXPLORER=on -DWITH_PARMYS=OFF -DWITH_ODIN=on"
```

Or something of that form. The warning seems to be produced when `odin_ii/src/verilog/verilog_bison.y` is parsed with Bison.

FYI. I think the reason you got those errors above is probably because you did not install the Titan benchmarks. They are very large and are not included in VTR directly. There are instructions on the wiki for how to install them properly.

AlexandreSinger, 2024-05-01:
After this PR has been merged: pr #2543 

Is this issue fully resolved?

amirarjmand93, 2024-05-02:
Yes. 
The warning about the Bison parser in Odin-ii that showcased **_2 shift/reduce conflicts [-Wconflicts-sr]_**  during building, has been resolved and disappeared.

closed 2024-05-16 by AlexandreSinger